ros-industrial/noether #2: Continous integration

VictorLamoine, 2017-05-09:
Take a look [here](https://github.com/InstitutMaupertuis/advanced_industrial_ci/
) at how to build your custom VTK version to be able to use VTK 7.1.0 in a CI environment.

I'm currently hosting VTK 7.1.0 binaries [here](http://victorlamoine.hopto.org/travis_ci/) (feel free to test with them!) but it won't last long since i'm moving from my flat and I won't have optical fiber anymore!

gavanderhoorn, 2017-05-09:
I have a feeling @AustinDeric might be on this / able to help. ros-industrial-consortium/A5#1.

VictorLamoine, 2017-05-09:
Yup I commented his pull request!

AustinDeric, 2017-05-09:
This is great thank you! 

All i did was build VTK 7.1 and PCL 1.8 manually in a docker image.  The problem is that its ~14 Gb in size and i am not sure why.  I never keep any of the data in the docker image.  Maybe VTK and PCL are that large but i find that hard to believe.  the rosindustrial/core image is about 2.5 Gb large.  

I am about to start pushing the Dockerfiles to https://github.com/ros-industrial/docker, any thoughts?

gavanderhoorn, 2017-05-09:
@AustinDeric wrote:
> I am about to start pushing the Dockerfiles to https://github.com/ros-industrial/docker, any thoughts?

I have some ideas, but without your `Dockerfile` it'd just be guesses.

gavanderhoorn, 2017-05-09:
Btw: I have Xenial debs for PCL 1.8 which are basically backports of the official Yakkety packages. They seem to work fine for me.

VictorLamoine, 2017-05-09:
@AustinDeric did you build in [MinSizeRel](https://cmake.org/cmake/help/v3.0/variable/CMAKE_BUILD_TYPE.html) mode with CMake? This ensures smaller (smallest?) binaries.
Also you can remove the sources after make install to spare some space (this is of course, less handy if you have to compile again)

On my machine:
- VTK 7.1.0 in Release mode weighs 423 Mb
- PCL 1.8.0 in Release mode weighs 454 Mb

I suggest using `ncdu` to trace what is eating disk space!

gavanderhoorn, 2017-05-09:
If anyone's interested: [xenial/pcl-1.8 debs](https://surfdrive.surf.nl/files/index.php/s/59c6msd9S52EhiK). IIRC these were compiled against the default VTK version for Xenial, so probably not 7.1.something.

AustinDeric, 2017-05-10:
Thank you for the help Gijs and Victor!

I got the [noether image](https://hub.docker.com/r/rosindustrial/noether/) image down to 4.15 Gb.  The VTK and PCL layer is 2.02 Gb.  [I used Gigj's cleanup code](https://github.com/ros-industrial/docker/pull/2/commits/977ed1ab58a282c7227eecaaa0552cd2fa8f8db9#diff-6df3936049c1c42e250e340492f8d893R11) for all the docker layers and Victor's MinSizeRel suggestion.

closed 2017-05-13 by VictorLamoine