ros-industrial/noether #10: node crashes for certain mesh

ipa-hsd, 2018-09-04:
I have tried to use this plugin with [godel](https://github.com/ros-industrial-consortium/godel) with success for "regular-shaped" surfaces (in this case a hemisphere). 

[image]

I tried it with an [arbitrary surface](https://ufile.io/jwy2q)  with varying results: 

[image]

[image]

The gdb output for (link for stl used mentioned above)
```
roslaunch noether noether.launch filename:=PATH_TO_STL_OR_PCD
```

```
Self intersection found
Self intersection found
Number of input cells for calculating intersection is less than 1.
Cannot compute intersection.
No intersection found for creating spline

Thread 1 "noether_node" received signal SIGSEGV, Segmentation fault.
0x00007ffff7537b77 in vtkAOSDataArrayTemplate<float>::GetTuple(long long) () from /usr/local/lib/libvtkCommonCore-7.1.so.1
(gdb) bt full
#0  0x00007ffff7537b77 in vtkAOSDataArrayTemplate<float>::GetTuple(long long) () from /usr/local/lib/libvtkCommonCore-7.1.so.1
No symbol table info available.
#1  0x00007ffff7ba542a in vtkPoints::GetPoint(long long) () from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#2  0x00007ffff7ba1f3e in tool_path_planner::RasterToolPathPlanner::resamplePoints(vtkSmartPointer<vtkPoints>&) () from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#3  0x00007ffff7b9f1b8 in tool_path_planner::RasterToolPathPlanner::checkPathForHoles(tool_path_planner::ProcessPath, std::vector<tool_path_planner::ProcessPath, std::allocator<tool_path_planner::ProcessPath> >&) () from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#4  0x00007ffff7b9cf03 in tool_path_planner::RasterToolPathPlanner::computePaths() () from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#5  0x00007ffff7b9c4ed in tool_path_planner::RasterToolPathPlanner::planPaths(vtkSmartPointer<vtkPolyData>, std::vector<tool_path_planner::ProcessPath, std::allocator<tool_path_planner::ProcessPath> >&)
    () from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#6  0x00007ffff7b9c602 in tool_path_planner::RasterToolPathPlanner::planPaths(std::vector<vtkSmartPointer<vtkPolyData>, std::allocator<vtkSmartPointer<vtkPolyData> > >, std::vector<std::vector<tool_path_planner::ProcessPath, std::allocator<tool_path_planner::ProcessPath> >, std::allocator<std::vector<tool_path_planner::ProcessPath, std::allocator<tool_path_planner::ProcessPath> > > >&) ()
   from /godel_ws/devel/lib/libraster_tool_path_planner.so
No symbol table info available.
#7  0x000000000040fba1 in main ()
No symbol table info available.
(gdb) 
```
I have gotten better results for PCL to mesh by using `greedy_projection` or `poisson` algorithms, but the tool paths look similar for each case. Any help is appreciated.

P.S.: Can anyone please update the maintainer email address? It does not function anymore. Thanks!

gavanderhoorn, 2018-09-04:
I would suggest to first compile all relevant libraries in debug mode. The backtrace is not very informative like this.

ipa-hsd, 2019-01-16:
Thanks @jrgnicho! I will try it out one of these days and get back.

ipa-hsd, 2019-02-15:
I created a dockerfile:

```
FROM osrf/ros:kinetic-desktop-full

RUN apt-get update \
  && apt-get install checkinstall wget

RUN mkdir /dev_noether && cd /dev_noether \
  && wget -q https://github.com/Kitware/VTK/archive/v7.1.0.tar.gz \
  && tar -xzf v7.1.0.tar.gz \
  && cd VTK-7.1.0 \
  && mkdir build \
  && cd build \
  && cmake ..
RUN cd /dev_noether/VTK-7.1.0/build \
  && make
RUN cd /dev_noether/VTK-7.1.0/build \
  && checkinstall --pkgname=vtk-7.1
 
RUN cd /dev_noether \
  && wget -q https://github.com/PointCloudLibrary/pcl/archive/pcl-1.8.0.tar.gz \
  && tar -xzf pcl-1.8.0.tar.gz

RUN echo "\
patch pcl-pcl-1.8.0/CMakeLists.txt <<'EOF' \n\
--- pcl-pcl-1.8.0/CMakeLists.txt    2019-02-14 14:36:38.114860268 +0100 \n\
+++ pcl-pcl-1.8.0/CMakeLists_7.txt    2019-02-14 12:44:54.652417038 +0100 \n\
@@ -360,5 +360,5 @@ \n\
 option(WITH_VTK "Build VTK-Visualizations" TRUE)\n\
 if(WITH_VTK AND NOT ANDROID)\n\
-  find_package(VTK)\n\
+  find_package(VTK 7.1 REQUIRED)\n\
   if(VTK_FOUND)\n\
     if(NOT DEFINED VTK_RENDERING_BACKEND)\n\
EOF" > /dev_noether/patch.sh \
  && cd /dev_noether \
  && chmod +x patch.sh \
  && ./patch.sh \
  && cd pcl-pcl-1.8.0 \
  && cmake . \
  && make 

RUN cd /dev_noether/pcl-pcl-1.8.0 \
  && checkinstall --pkgname=pcl-1.8

RUN mkdir -p /noether_ws/src && cd /noether_ws/src \
  && git clone --depth 1 https://github.com/ros-industrial/noether.git -b master \
  && cd /noether_ws \
  && apt-get update -qq && rosdep install -i -y --from-paths src \
  && /opt/ros/kinetic/env.sh catkin_make -DCMAKE_BUILD_TYPE=Debug
  
RUN apt-get update -qq && apt-get install -qq gdb

ENTRYPOINT ["/noether_ws/devel/env.sh"]
```
And ran the docker image:

```
docker run --rm --privileged -v /home/user/noether/meshes:/meshes --name noether --net=host -e ROS_NAMESPACE modified_noether:latest gdb -ex run /noether_ws/devel/lib/noether/surface_raster_planner_application _filename:=/meshes/rock.stl
```
I get the following bt:
```
(gdb) bt
#0  0x00007ffff4956428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff495802a in __GI_abort () at abort.c:89
#2  0x00007ffff5a8b99f in vtkXOpenGLRenderWindow::GetDesiredVisualInfo (this=0xac0fd0) at /dev_noether/VTK-7.1.0/Rendering/OpenGL2/vtkXOpenGLRenderWindow.cxx:302
#3  0x00007ffff5a8efea in vtkXOpenGLRenderWindow::GetDesiredDepth (this=0xac0fd0) at /dev_noether/VTK-7.1.0/Rendering/OpenGL2/vtkXOpenGLRenderWindow.cxx:1223
#4  0x00007ffff5a7ccb2 in vtkXRenderWindowInteractor::Initialize (this=0x2c404d0) at /dev_noether/VTK-7.1.0/Rendering/OpenGL2/vtkXRenderWindowInteractor.cxx:342
#5  0x00007fffeb772fdf in vtkRenderWindow::Render (this=0xac0fd0) at /dev_noether/VTK-7.1.0/Rendering/Core/vtkRenderWindow.cxx:317
#6  0x00007ffff59c1eb4 in vtkOpenGLRenderWindow::Render (this=0xac0fd0) at /dev_noether/VTK-7.1.0/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx:692
#7  0x00007ffff5a91a1c in vtkXOpenGLRenderWindow::Render (this=0xac0fd0) at /dev_noether/VTK-7.1.0/Rendering/OpenGL2/vtkXOpenGLRenderWindow.cxx:1856
#8  0x00007ffff6d5cd47 in vtk_viewer::VTKViewer::renderDisplay (this=0x7fffffffe270) at /noether_ws/src/noether/vtk_viewer/src/vtk_viewer.cpp:54
#9  0x00000000004113b6 in noether::SurfaceRasterPlannerApplication::visualizeDisplay (this=0x7fffffffe270) at /noether_ws/src/noether/noether/include/noether/surface_raster_planner_application.h:57
#10 0x0000000000410405 in main (argc=1, argv=0x7fffffffea18) at /noether_ws/src/noether/noether/src/surface_raster_planner_application.cpp:229
```
Now the error is in visualization, and no more in path planning.

P.S.: Is this new version compatible with [godel_noether](https://github.com/Jmeyer1292/godel_noether), since the executables have changed in this repo.

ipa-hsd, 2019-02-20:
I tried it with godel, and this is the error I receive (even though there is no crash anymore):

```
[ERROR] [RasterPathPlanner](L:805): Number of input cells for calculating intersection is less than 1, cannot compute intersection
[ERROR] [RasterPathPlanner](L:805): Number of input cells for calculating intersection is less than 1, cannot compute intersection
[INFO ] [RasterPathPlanner](L:271): Deleting path 2
[INFO ] [RasterPathPlanner](L:271): Deleting path 2
```
Are there any parameters to be set to cover the entire surface? Or I need to choose a simpler object?
Below is how the visualization of the path:

[image]