ros-industrial/noether #14: Update Maintainer and License information

mpowelson, 2018-12-04:
The license I'm told should actually be Apache 2.0. I suppose @jrgnicho , you are the de facto maintainer now?

jrgnicho, 2019-01-15:
Other repos under the ros-industrial organization are using BSD as well.  I'm not sure on this one but I think BSD is the more flexible one.

gavanderhoorn, 2019-01-15:
IANAL, but both BSD and Apache 2 are pretty similar in terms of 'flexibility' iiuc.

The difference is that Apache 2 has a patent clause, which offers better protection.

For generic ROS-Industrial packages, Apache 2 is recommended.

jrgnicho, 2019-01-15:
@gavanderhoorn thanks for clarifying that.  @mpowelson we probably want to use Apache2 then.

mpowelson, 2020-07-06:
I believe license was fixed during wrid19. This year maybe we can finally knock out maintainer info.

mpowelson, 2020-07-06:
@jrgnicho Are you the maintainer of this repo?

mpowelson, 2020-07-07:
@jrgnicho I'm going to assume yes. I am also going to use your gmail devel account. That is what you used on noether_filtering

mpowelson, 2020-07-07:
Addressed in #92

closed 2020-07-07 by mpowelson