ros-industrial/noether #20: Update the ci docker image

jrgnicho, 2019-01-16:
The CI currently fails due to a dependency on [pcl_conversions](https://travis-ci.org/ros-industrial/noether/jobs/470109028).  The PR #16 introduced changes that removed the ros pcl_conversions dependency as keeping it there would cause runtime errors.  I can build the latest code in a machine with ROS kinetic, PCL 1.8 and VTK 7.1. This suggest to me that the build error is due to the current configuration of the noether docker container.

ipa-hsd, 2019-02-15:
If you have not built the docker image already, I have done it so https://github.com/ros-industrial/noether/issues/10#issuecomment-464087967. Should I upload the docker image?

jrgnicho, 2019-02-15:
We found out that our build configuration was the root cause of the problem.  This problem was fixed in #25; thanks for tracking this issue however.

closed 2019-02-15 by jrgnicho