ros-industrial/noether #36: Error: accumulate is not a member of 'std'

jclaros92, 2019-03-27:
I followed the readme and installed the VTK and PCL libraries but now this error about 'accumulate' not being apart of 'std' is occurring. It seems like an include may be missing in one of the PCL files or something was installed improperly.

/src/noether/tool_path_planner/src/raster_path_generator.cpp: In member function ‘boost::optional<std::vector<geometry_msgs::PoseArray_<std::allocator<void> > > > tool_path_planner::RasterPathGenerator::generate(const tool_path_planner::ProcessTool&, const pcl::PolygonMesh&) const’:
/home/rosindustrial1/catkin_ws/src/noether/tool_path_planner/src/raster_path_generator.cpp:321:38: error: ‘accumulate’ is not a member of ‘std’
   std::size_t non_empty_paths = std::accumulate(tool_process_paths.begin(), tool_process_paths.end(),0,[](
                                      ^~~~~~~~~~
make[2]: *** [CMakeFiles/tool_path_planner.dir/src/raster_path_generator.cpp.o] Error 1
make[1]: *** [CMakeFiles/tool_path_planner.dir/all] Error 2
make: *** [all] Error 2

Any ideas on how to fix this error?

jrgnicho, 2019-03-29:
Are you building this in kinetic?

jclaros92, 2019-04-10:
No, my current build is in melodic. Is this not ready for melodic?

gavanderhoorn, 2019-04-10:
@jrgnicho: `std::accumulate(..)` is defined in the `numeric` header. Is that included?

jrgnicho, 2019-04-10:
@jclaros92  We have been using this branch in ros-kinetic Ubuntu 16.04.  
@gavanderhoorn it is possible that the __numeric__ header isn't included and in melodic it isn't being added indirectly by another header file.

jclaros92, 2019-04-17:
@jrgnicho @gavanderhoorn Any updates on this error? Still unable to build this library.

gavanderhoorn, 2019-04-17:
> @jrgnicho @gavanderhoorn Any updates on this error? Still unable to build this library.

well, have you tried adding the `numeric` header include?

Levi-Armstrong, 2019-04-17:
I am using melodic and you do need to add the include <numeric> as @gavanderhoorn stated to file using std::accumulate.

gavanderhoorn, 2019-04-18:
See #38.

jclaros92, 2019-04-18:
I appreciate everyone's support. I'm new to working with packages in this environment so this simple error had me stumped. Thanks again.

jrgnicho, 2019-04-19:
#38 has been merged.  It should build in melodic now

closed 2019-05-15 by jrgnicho

DechaoZeng, 2021-06-02:
> @jrgnicho @gavanderhoorn Any updates on this error? Still unable to build this library.

Any update on this problem? I am working on installing PCL libraries as well. Appreciate all!

gavanderhoorn, 2021-06-02:
@DechaoZeng: what sort of update are you looking for?

The issue the OP encountered was solved in #38. It also wasn't a problem with install PCL, but a missing header.