ros-industrial/noether #39: Error running surf_raster_planner_application.launch

jclaros92, 2019-04-23:
I'm trying to run the **surf_raster_planner_application.launch** file and the node keeps crashing without any feedback. I added some print outs in the **surface_raster_planner_application.cpp** and noticed it keeps crashing at these lines:
`  tool_path_planner::ProcessTool tool = loadTool(pnh);
  tool_path_planner::RasterToolPathPlanner planner;
  planner.enableConsoleDebug(console_debug_on);`
I'm unsure of why the node keeps crashing after these lines. The config folder contains the tool.yaml file.

One concern I have is that when I built the noether package I got the following warnings,

**CMake Warning at /home/ros3/git_workspace/noether_test/src/noether/noether/CMakeLists.txt:43 (add_executable):
  Cannot generate a safe runtime search path for target
  surface_raster_planner_server because files in some directories may
  conflict with libraries in implicit directories:

    runtime library [libpcl_common.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_octree.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_io.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at /home/ros3/git_workspace/noether_test/src/noether/noether/CMakeLists.txt:59 (add_executable):
  Cannot generate a safe runtime search path for target segmentation_server
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpcl_common.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_octree.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_io.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at /home/ros3/git_workspace/noether_test/src/noether/noether/CMakeLists.txt:51 (add_executable):
  Cannot generate a safe runtime search path for target
  surface_raster_planner_application because files in some directories may
  conflict with libraries in implicit directories:

    runtime library [libpcl_common.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_octree.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libpcl_io.so.1.8] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.**

Has anyone else encountered an issue when running this launch file?

jclaros92, 2019-04-23:
I also tried running the **mesh_segmenter_node.launch** file and received the error shown below. It seems that its pulling from the wrong version of VTK even though I installed the 7.1 version. I must be missing something in one of my CMakeLists.txt.

/home/ros3/sample_data/sphere.stl
ERROR: In /build/vtk6-VHOYAG/vtk6-6.3.0+dfsg1/IO/Geometry/vtkSTLReader.cxx, line 97
vtkSTLReader (0x55b567790840): A FileName must be specified.

ERROR: In /build/vtk6-VHOYAG/vtk6-6.3.0+dfsg1/Common/ExecutionModel/vtkExecutive.cxx, line 784

jrgnicho, 2019-04-23:
What's the default version of PCL that's installed in your system?

jclaros92, 2019-04-24:
PCL 1.8. I followed the Readme and installed VTK 7.1 and PCL 1.8. It seems as the VTK library installed all the *.so files in /usr/local/lib/ and the noether package is looking for them in /usr/lib/x86_64-linux-gnu/.

jrgnicho, 2019-04-24:
During the PCL library installation did you follow the:
> Locate the find package(VTK) line (close to line 362) and edit it to find_package(VTK 7.1 REQUIRED)

Step? 

Also, clean your catkin workspace `catkin clean -y` if you haven't done so

jclaros92, 2019-04-24:
I double checked and I did add the "VTK 7.1 REQUIRED" line. I have tried cleaning and that has not fixed my issue.

jrgnicho, 2019-05-15:
All I can suggest is to  check your PATH env variable, it may be introducing paths to alternate versions of the VTK libray