ros-industrial/noether #52: Why is VKT 7.1 required?

Levi-Armstrong, 2019-08-13:
I noticed that on melodic it comes with PCL 1.8.1 and VTK 6.3 and curious why version 7.1 is required.

DavidMerzJr, 2019-10-28:
libvtk7-dev is also available on Ubuntu 18.04/ROS Melodic, which supplies VTK 7.1.  Does this work for building earlier commits (e.g. commit a82ed0a7b96d5ac688462069759058144b1c895e)?

However, now the requirements have been changed to use PCL 1.9.1 and VTK 8.2.  I understand that PCL 1.9.1 is required for NURBS fitting.  Do we really need VTK 8.2, or could we go back to VTK 7.1?  It seems that staying on a VTK version available through apt would make the Noether library much easier to install and use.

jrgnicho, 2019-10-29:
@Levi-Armstrong I don't recall what specific feature of VTK 7.1 was required but whatever that feature was it wasn't available on previous versions
@DavidMerzJr  VTK 8.2 gets rid of deprecated warnings that get printed during segmentation, other than that it is not required.  We could make the compiled debians for PCL and VTK available in order to facilitate installation.  It's just a matter of finding a proper way to host those files

closed 2019-12-02 by Levi-Armstrong