ros-industrial/noether #71: Propose adding Clang-Tidy

Levi-Armstrong, 2019-12-02:
@jrgnicho Below is a macro used for adding clang-tidy to a target if interested in adding it to noether. It may need to be modified because it run only if ENABLE_CLANG_TIDY OR ENABLE_TESTS variable is set to on which are tesseract variables.

```
# Add clang-tidy to a target if ENABLE_CLANG_TIDY or ENABLE_TESTS is enabled
# Usage: noether_clang_tidy(Target) or noether_clang_tidy(Target true) or noether_clang_tidy(Target false)
#    * noether_clang_tidy(Target) adds clang tidy with warnings as errors
#    * noether_clang_tidy(Target true) adds clang tidy with warnings as errors
#    * noether_clang_tidy(Target false) adds clang tidy with warnings as warnings
macro(noether_clang_tidy target)
  cmake_parse_arguments(ARG "true,false" "" "" ${ARGN})

  get_target_property(${target}_type ${target} TYPE)

  # Add clang tidy
  if (NOT ${${target}_type} STREQUAL "INTERFACE_LIBRARY")
    if (ENABLE_CLANG_TIDY OR ENABLE_TESTS)
      find_program(CLANG_TIDY_EXE NAMES "clang-tidy" DOC "Path to clang-tidy executable")
      if(NOT CLANG_TIDY_EXE)
        message(WARNING "clang-tidy not found.")
      else()
        message(STATUS "clang-tidy found: ${CLANG_TIDY_EXE}")
        if(ARG_false)
          set(DO_CLANG_TIDY "${CLANG_TIDY_EXE}" "-checks=-*,bugprone-*,cppcoreguidelines-avoid-goto,cppcoreguidelines-c-copy-assignment-signature,cppcoreguidelines-interfaces-global-init,cppcoreguidelines-narrowing-conversions,cppcoreguidelines-no-malloc,cppcoreguidelines-slicing,cppcoreguidelines-special-member-functions,misc-*,modernize-*,performance-*,readability-avoid-const-params-in-decls,readability-container-size-empty,readability-delete-null-pointer,readability-deleted-default,readability-else-after-return,readability-function-size,readability-identifier-naming,readability-inconsistent-declaration-parameter-name,readability-misleading-indentation,readability-misplaced-array-index,readability-non-const-parameter,readability-redundant-*,readability-simplify-*,readability-static-*,readability-string-compare,readability-uniqueptr-delete-release,readability-rary-objects")
        else()
          set(DO_CLANG_TIDY "${CLANG_TIDY_EXE}" "-checks=-*,bugprone-*,cppcoreguidelines-avoid-goto,cppcoreguidelines-c-copy-assignment-signature,cppcoreguidelines-interfaces-global-init,cppcoreguidelines-narrowing-conversions,cppcoreguidelines-no-malloc,cppcoreguidelines-slicing,cppcoreguidelines-special-member-functions,misc-*,modernize-*,performance-*,readability-avoid-const-params-in-decls,readability-container-size-empty,readability-delete-null-pointer,readability-deleted-default,readability-else-after-return,readability-function-size,readability-identifier-naming,readability-inconsistent-declaration-parameter-name,readability-misleading-indentation,readability-misplaced-array-index,readability-non-const-parameter,readability-redundant-*,readability-simplify-*,readability-static-*,readability-string-compare,readability-uniqueptr-delete-release,readability-rary-objects" "-warnings-as-errors=-*,bugprone-*,cppcoreguidelines-avoid-goto,cppcoreguidelines-c-copy-assignment-signature,cppcoreguidelines-interfaces-global-init,cppcoreguidelines-narrowing-conversions,cppcoreguidelines-no-malloc,cppcoreguidelines-slicing,cppcoreguidelines-special-member-functions,misc-*,modernize-*,performance-*,readability-avoid-const-params-in-decls,readability-container-size-empty,readability-delete-null-pointer,readability-deleted-default,readability-else-after-return,readability-function-size,readability-identifier-naming,readability-inconsistent-declaration-parameter-name,readability-misleading-indentation,readability-misplaced-array-index,readability-non-const-parameter,readability-redundant-*,readability-simplify-*,readability-static-*,readability-string-compare,readability-uniqueptr-delete-release,readability-rary-objects")
        endif()
        set_target_properties("${target}" PROPERTIES CXX_CLANG_TIDY "${DO_CLANG_TIDY}")
      endif()
    endif()
  endif()
endmacro()

```

mpowelson, 2020-07-06:
More broadly, revamp the cmake to use https://github.com/ros-industrial/cmake_common_scripts

mpowelson, 2020-07-07:
I'm starting on this, but I'd say it is unlikely that I will get all of the packages done.