ros-industrial/noether #94: roslaunch noether_examples surf_raster_planner_demo.launch error

nevermore0127, 2020-07-10:
The problem is as below ,when i run :roslaunch noether_examples surf_raster_planner_demo.launch
[ERROR] [RasterPathPlanner](L:980): No connected lines were found
[INFO ] [RasterPathPlanner](L:281): Deleting 3 paths
[ INFO] [1594391522.205637207]: log directory /home/xtark/.ros/log/14255b0c-c2ba-11ea-9c3c-000c296c4647
================================================================================REQUIRED process [surface_raster_planner_demo-2] has died!
process has died [pid 101370, exit code -11, cmd /home/xtark/catkin_hyy/devel/lib/noether/surface_raster_planner_application __name:=surface_raster_planner_demo __log:=/home/xtark/.ros/log/14255b0c-c2ba-11ea-9c3c-000c296c4647/surface_raster_planner_demo-2.log].
log file: /home/xtark/.ros/log/14255b0c-c2ba-11ea-9c3c-000c296c4647/surface_raster_planner_demo-2*.log
Initiating shutdown!
================================================================================

it seems some thing is wrong whit getConnectedIntersectionLine? what can i do for this ? thanks

closed 2020-07-11 by nevermore0127

Beshario, 2020-09-21:
I have the same issue, I will pass my command line messages
Do you mind sharing how you solved it?:
```
asp@asp-MS-7B84:~/catkin_ws$ roslaunch noether_examples surf_raster_planner_demo.launch
... logging to /home/asp/.ros/log/b8e2f5a0-fc3f-11ea-9ab7-d037451d0501/roslaunch-asp-MS-7B84-15530.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://asp-MS-7B84:40433/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /surface_raster_planner_demo/debug_on: False
 * /surface_raster_planner_demo/filename: /home/asp/catkin_...
 * /surface_raster_planner_demo/intersecting_plane_height: 0.1
 * /surface_raster_planner_demo/line_spacing: 0.15
 * /surface_raster_planner_demo/min_hole_size: 0.01
 * /surface_raster_planner_demo/min_segment_size: 0.01
 * /surface_raster_planner_demo/pt_spacing: 0.05
 * /surface_raster_planner_demo/raster_angle: 0.0
 * /surface_raster_planner_demo/raster_wrt_global_axes: False
 * /surface_raster_planner_demo/tool_offset: 0.0

NODES
  /
    surface_raster_planner_demo (noether/surface_raster_planner_application)

auto-starting new master
process[master]: started with pid [15540]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b8e2f5a0-fc3f-11ea-9ab7-d037451d0501
process[rosout-1]: started with pid [15551]
started core service [/rosout]
process[surface_raster_planner_demo-2]: started with pid [15558]
Error:   No connected lines were found
         at line 965 in /home/asp/catkin_ws/src/noether/tool_path_planner/src/raster_tool_path_planner.cpp
Info:    Deleting 3 paths
[ INFO] [1600716133.213281909]: log directory /home/asp/.ros/log/b8e2f5a0-fc3f-11ea-9ab7-d037451d050
```

Beshario, 2020-09-23:
I am guessing this may not be a problem, moving forward