ros-industrial/noether #96: Generating path plans for specific areas on a mesh

madhan001, 2020-07-26:
Is there a way to generate path plans for specific areas on a mesh instead of the entire mesh ?

closed 2020-11-17 by madhan001

Levi-Armstrong, 2020-11-17:
There are some segmentation tools and I believe there is a tool for drawing on a mesh and extracting that section then we send it to noether. @jrgnicho and @marrts Is the draw tool currently available?

drchrislewis, 2020-11-17:
@madhan001 the swri-robotics toolpath_offline_planner has the tools you need. It is being rapidly revised. However as it is should work. One should first segment the part, then select a region, then generate your paths. One of the PR's uses the latest noether interface. The others need to use a tagged version. Don't bother with the heat method at this point it is finicky but works extremely well on error free meshes. It can find paths completely covering very complex surfaces or any selected region.

madhan001, 2020-11-18:
Thanks for the recommendation, I'll look into it