ros-industrial/noether #98: CMake Error: find package "vtk_viewer"

IKapitaniuk, 2020-07-28:
After the last merge, aka "Update vtk_viewer CMake #93" the CMake error has appeared:
```
Errors     << mesh_segmenter:cmake /catkin_ws/logs/mesh_segmenter/build.cmake.000.log                                                                                                    
CMake Error at /catkin_ws/src/noether/mesh_segmenter/CMakeLists.txt:16 (find_package):
  By not providing "Findvtk_viewer.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "vtk_viewer", but CMake did not find one.

  Could not find a package configuration file provided by "vtk_viewer" with
  any of the following names:

    vtk_viewerConfig.cmake
    vtk_viewer-config.cmake

  Add the installation prefix of "vtk_viewer" to CMAKE_PREFIX_PATH or set
  "vtk_viewer_DIR" to a directory containing one of the above files.  If
  "vtk_viewer" provides a separate development package or SDK, be sure it has
  been installed.

cd /catkin_ws/build/mesh_segmenter; catkin build --get-env mesh_segmenter | catkin env -si  /usr/bin/cmake /catkin_ws/src/noether/mesh_segmenter --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/catkin_ws/devel/.private/mesh_segmenter -DCMAKE_INSTALL_PREFIX=/catkin_ws/install; cd -
```

Frankly speaking, I have no idea how exactly catkin build works under the hood, but I found that the problem can be solved by adding the following lines into package.xml:
```
<export>
    <build_type>cmake</build_type>
  </export>
```
but I am not sure that this is a proper solution.

Levi-Armstrong, 2020-07-28:
That was the correct fix. Do you mind creating a PR?

IKapitaniuk, 2020-07-28:
Another problem, not really relevant to the one mentioned above but also regarding the CMake behavior, is that the very first build after the cleaning of the catkin environment crashes with the following error:
```
Errors     << noether_filtering:cmake /catkin_ws/logs/noether_filtering/build.cmake.000.log                                                                                              
CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

CMake Error:
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:CXX>

  $<COMPILE_LANGUAGE:...> may only be used to specify include directories,
  compile definitions, compile options, and to evaluate components of the
  file(GENERATE) command.

PCL NURBS FOUND
Building mesh plugins
cd /catkin_ws/build/noether_filtering; catkin build --get-env noether_filtering | catkin env -si  /usr/bin/cmake /catkin_ws/src/noether/noether_filtering --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/catkin_ws/devel; cd -
```
All the following executions of the "catkin build' are successful. It looks like some missing information is added after the very first build attempt.

Levi-Armstrong, 2020-07-28:
@mpowelson Does this look familiar?

IKapitaniuk, 2020-07-28:
> That was the correct fix. Do you mind creating a PR?

Sure

> @mpowelson Does this look familiar?

I see that for quite a long time, but it annoys me only the very first time, that's why I didn't care too much.

mpowelson, 2020-07-28:
Not particularly. What version of cmake are you using? Are you on xenial/kinetic?

IKapitaniuk, 2020-07-28:
I use Docker for testing both kinetic and melodic. It happens on both platforms.
The base images are _ros:melodic-ros-base_ and _ros:melodic-ros-base_ the latest available. 
The dedicated workspace for Noether. The dependency resolved automatically via wstool and rosdep install.
Deb packages vtk-7.1.1 and pcl-1.9.1 are made using the instructions in README.md and installed via dpkg -i <pkg>.deb
That's it.

Cmake versions:
melodic - cmake 3.10.2
kinetic - cmake version 3.5.1

Levi-Armstrong, 2020-07-29:
I would upgrade the package to leverage cmake_common_scripts and see if the issue goes away. You can use vtk_viewer as a guide.

dave992, 2020-07-30:
I have the same issue (without using Docker) on Ubuntu 18.04/ROS Melodic where the build fails on the `noether_filtering` package the very first time building after cleaning the workspace. 

To reproduce:
* Install PCL and VTK based on the instructions in the readme
* Install other dependencies using `rosdep` and `wstool`
* If workspace is not cleaned: `catkin clean -y`
* `catkin build`

Any subsequent builds will then be successful until you clean the workspace again.

gavanderhoorn, 2020-07-30:
@dave992: #97 doesn't fix this?

Levi-Armstrong, 2020-07-30:
Try commenting out the lines below in noether_filtering cmake file and see if the issue goes away on first build.

```
if(NOT DEFINED NURBS_FOUND)
    try_compile(
        NURBS_FOUND
        ${CMAKE_CURRENT_BINARY_DIR}/pcl_nurbs_try_compile
        ${CMAKE_CURRENT_SOURCE_DIR}/cmake/test_scripts/check_pcl_nurbs.cpp
        CMAKE_FLAGS ${CMAKE_FLAGS} -DINCLUDE_DIRECTORIES=${PCL_INCLUDE_DIRS}
        LINK_LIBRARIES ${PCL_LIBRARIES}
        )
    if (NOT ${NURBS_FOUND})
        message("PCL NURBS NOT FOUND")
    else()
        message("PCL NURBS FOUND")
    endif()
endif()
```

dave992, 2020-08-03:
That indeed solves the error on the first build, I'll make a PR for this then.

Levi-Armstrong, 2020-08-03:
That is good news, but it will not be as simple to just remove it. This was added as a way to check if PCL was built with NURBS so an alternative method needs to setup. A quick fix would be to add a compiler flag ENABLE_NURBS which is defaulted to OFF since that is the default of PCL.

```
if (not ENABLE_NURBS)
   set(ENABLE_NURBS, OFF)
endif()
```

then switch NURBS_FOUND for ENABLE_NURBS everywhere in the cmake

Levi-Armstrong, 2020-08-07:
@IKapitaniuk  and @dave992 it turns out there is a bug in cmake which is fixed in noetic. If you pip install cmake this will get you the latest version. If that is not desirable you can pass the cmake compiler flag -DNURBS_FOUND=ON/OFF depending on if your version of PCL was built with NURBS.

dave992, 2020-08-10:
As this issue indeed seems specific for 18.04/melodic, passing the -DNURBS_FOUND flag sounds like an acceptable solution to me. I'll use the cmake compiler flag for now then. Thanks for the assistance! 

P.S. I don't know how many people using Noether are still on melodic, but do you want me to add this instruction to the readme?