ros-industrial/noether #103: Inconsistent generated toolpaths using surface_raster_planner_demo.launch

dave992, 2020-08-14:
Based on my understanding of Noether, it should generate a toolpath in a deterministic way. Therefore recalculating the toolpath should always return the same result. 

When running the `surface_raster_planner_demo.launch` the results vary, here are some screenshots of what I mean:
| | |
|:-------------------------:|:-------------------------:|
|[image]|[image]|
|[image]|[image]|

The 'main' toolpaths seem to be recalculated correctly, but some additional clusters seem to appear/disappear on different runs. Also, some runs seem to result in a single huge arrow for the example mesh. Using another mesh also results in this behavior. 

To reproduce (It may take several tries to produce different results):
1. `roslaunch noether_examples surface_raster_planner_demo.launch`
2. Close window or kill process
3. Repeat

This was run on Ubuntu 18.04/ROS Melodic with VTK 7.1.1 and PCL 1.9.1.

gavanderhoorn, 2020-08-14:
@dave992: could you perhaps also specify what version of Noether you are seeing this with (ie: commit hash)?

dave992, 2020-08-14:
Both PC's I encountered this issue on were using version 47d74720252bc724e540e9819be61bba2718758f.

gavanderhoorn, 2020-08-25:
@Levi-Armstrong: is this something you've ran into perhaps?

We're willing to debug this ourselves, just looking to avoid spending effort in case it's a known issue.

Levi-Armstrong, 2020-08-25:
@jrgnicho @DavidMerzJr @drchrislewis  you all probably have used it the most. Do you have any insights into what they are experiencing? My first thoughts are it should be, at least for the first raster, but then it does some offsetting to create equal surface distance lines which may not be deterministic.

gavanderhoorn, 2020-08-25:
We were also thinking it may be using some uninitialised memory somewhere. Especially the screenshot in the bottom-right made me suspicious.

mpowelson, 2020-08-25:
> We were also thinking it may be using some uninitialised memory somewhere. Especially the screenshot in the bottom-right made me suspicious.

I remember thinking the same thing wrid19 when we had some unit tests that would pass on one run and fail the next if ran repeatedly. Probably unrelated, but [here](https://github.com/ros-industrial/noether/issues/47) is the issue