ros-industrial/noether #104: Build errors on Ubuntu 18.04 + ROS Melodic

shubhamwagh, 2020-09-09:
I am currently trying to install this package. I did follow the instructions starting with VTK and PCL installations.

After `catkin build` in my workspace which contains `src/noether` packages, my build fails.
```
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/qbot/toolpath
--------------------------------------------------------
Build Space:        [exists] /home/qbot/toolpath/build
Devel Space:        [exists] /home/qbot/toolpath/devel
Install Space:      [unused] /home/qbot/toolpath/install
Log Space:         [missing] /home/qbot/toolpath/logs
Source Space:       [exists] /home/qbot/toolpath/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                         
[build] Updating package table.                                                                    
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                
Finished  <<< catkin_tools_prebuild                [ 1.1 seconds ]                                 
Starting  >>> noether_filtering                                                                    
Starting  >>> noether_msgs                                                                         
Starting  >>> vtk_viewer                                                                           
___________________________________________________________________________________________________
Errors     << vtk_viewer:cmake /home/qbot/toolpath/logs/vtk_viewer/build.cmake.000.log             
CMake Error at /home/qbot/toolpath/src/noether/vtk_viewer/CMakeLists.txt:4 (find_package):
  By not providing "Findcmake_common_scripts.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cmake_common_scripts", but CMake did not find one.

  Could not find a package configuration file provided by
  "cmake_common_scripts" with any of the following names:

    cmake_common_scriptsConfig.cmake
    cmake_common_scripts-config.cmake

  Add the installation prefix of "cmake_common_scripts" to CMAKE_PREFIX_PATH
  or set "cmake_common_scripts_DIR" to a directory containing one of the
  above files.  If "cmake_common_scripts" provides a separate development
  package or SDK, be sure it has been installed.

cd /home/qbot/toolpath/build/vtk_viewer; catkin build --get-env vtk_viewer | catkin env -si  /usr/local/bin/cmake /home/qbot/toolpath/src/noether/vtk_viewer --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/qbot/toolpath/devel; cd -
...................................................................................................
Failed     << vtk_viewer:cmake                     [ Exited with code 1 ]                          
Failed    <<< vtk_viewer                           [ 0.5 seconds ]                                 
Abandoned <<< noether_conversions                  [ Unrelated job failed ]                        
Abandoned <<< mesh_segmenter                       [ Unrelated job failed ]                        
Abandoned <<< tool_path_planner                    [ Unrelated job failed ]                        
Abandoned <<< path_sequence_planner                [ Unrelated job failed ]                        
Abandoned <<< noether                              [ Unrelated job failed ]                        
Abandoned <<< noether_examples                     [ Unrelated job failed ]                        
Finished  <<< noether_msgs                         [ 3.4 seconds ]                                 
___________________________________________________________________________________________________
Warnings   << noether_filtering:cmake /home/qbot/toolpath/logs/noether_filtering/build.cmake.000.log
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** 2d features related to ensenso will be disabled
** WARNING ** 2d features related to davidSDK will be disabled
** WARNING ** 2d features related to dssdk will be disabled
** WARNING ** 2d features related to pcap will be disabled
** WARNING ** 2d features related to png will be disabled
** WARNING ** 2d features related to libusb-1.0 will be disabled
CMake Warning at /home/qbot/toolpath/src/noether/noether_filtering/CMakeLists.txt:15 (message):
  The minimum required version of PCL is 1.9, but found 1.8.1 in path first.
  Checking for exactly 1.9

CMake Warning (dev) at CMakeLists.txt:16 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable PCL_ROOT is set to:

    /usr/local

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** 2d features related to ensenso will be disabled
** WARNING ** 2d features related to davidSDK will be disabled
** WARNING ** 2d features related to dssdk will be disabled
** WARNING ** 2d features related to pcap will be disabled
** WARNING ** 2d features related to png will be disabled
** WARNING ** 2d features related to libusb-1.0 will be disabled
PCL NURBS NOT FOUND
Building mesh plugins
cd /home/qbot/toolpath/build/noether_filtering; catkin build --get-env noether_filtering | catkin env -si  /usr/local/bin/cmake /home/qbot/toolpath/src/noether/noether_filtering --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/qbot/toolpath/devel; cd -
...................................................................................................
Finished  <<< noether_filtering                    [ 24.4 seconds ]                                
[build] Summary: 3 of 10 packages succeeded.                                                       
[build]   Ignored:   None.                                                                         
[build]   Warnings:  1 packages succeeded with warnings.                                           
[build]   Abandoned: 6 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                            
[build] Runtime: 25.6 seconds total.                      
```

May I know what went wrong? Thanks.

shubhamwagh, 2020-09-09:
It will better to add extra steps for installation for people who are not aware about rosinstall file dependencies.

1) Initialize workspace using given rosintsall file
`wstool init src ./src/noether/dependencies_ros1.rosinstall`

2) if `lcov` is not installed: `sudo apt-get install lcov`
3) `catkin build`

gavanderhoorn, 2020-09-09:
I've re-opened this, as the readme does indeed not list the steps which would populate the workspace.

Levi-Armstrong, 2020-09-09:
@gavanderhoorn The procedure in the previous comment is what I would expect someone to do when pulling a repo they have never used; should we add the wstool procedure to the readme?

gavanderhoorn, 2020-09-09:
> The procedure in the previous comment is what I would expect someone to do when pulling a repo they have never used;

Yes, I agree with this, and in general I would say if someone has some experience in populating workspaces with ROS packages they should have some understanding of all this.

However, there are a sufficient nr of users for which these kinds of readmes are their first "foray" into building packages from source, so if we want to be kind to users we should probably include a short section with the `wstool` (or `vcstool`) commands they could use.

New users will copy-paste them.

More experienced users will just ignore them (or perhaps also copy-paste).

Levi-Armstrong, 2020-09-09:
Sounds good to me.

Levi-Armstrong, 2020-09-09:
Should we be transitioning to [vcstool](http://wiki.ros.org/vcstool)?

gavanderhoorn, 2020-09-09:
You could.

There's no immediate actual *need* though I believe.

As you can see, the set of commands would be relatively similar.

Levi-Armstrong, 2020-09-09:
Update the readme build section with the following?

## Build
- Create a catkin workspace and clone this repository in the `src` directory
- Navigate to workspace directory
- Import dependencies
  - `vcstool import src < src/noether/dependencies_ros1.rosinstall`
  - `vcstool pull src`
- Build the noether packages
  ```
  catkin build noether
  ```
  
  _Note: You can just run `catkin build` in order to build everything in your workspace, including the noether packages_

gavanderhoorn, 2020-09-09:
Yes, that seems like a nice overview.

A bit more terse:

```bash
# if you already have a workspace, skip this step
mkdir -p ~/catkin_ws/src

# navigate to the root of your workspace
cd ~/catkin_ws

# pull down the dependencies
vcstool import src/ < src/noether/dependencies_ros1.rosinstall
vcstool pull src/

# build the Noether packages
# note: if you want to build the entire workspace, just run 'catkin build' instead
catkin build noether
```

Levi-Armstrong, 2020-09-09:
> I don't believe vcstool pull src is needed for an initial import btw.

good to know. The reason I asked about vcstool is I noticed that wstool repo has been archived.

Levi-Armstrong, 2020-09-09:
Should we add the clone repository step?

```
# if you already have a workspace, skip this step
mkdir -p ~/catkin_ws/src

# if you already cloned noether, skip this step
cd ~/catkin_ws/src && git clone https://github.com/ros-industrial/noether.git

# navigate to the root of your workspace
cd ~/catkin_ws

# pull down the dependencies
vcstool import src < src/noether/dependencies_ros1.rosinstall

# build the Noether packages
# note: if you want to build the entire workspace, just run 'catkin build' instead
catkin build noether
```

marip8, 2021-06-02:
Closing; addressed by #106

closed 2021-06-02 by marip8