ros-industrial/noether #108: Mesh Filtering process died on launching.

shubhamwagh, 2020-09-15:
Hi,

On launching mesh_filtering launch file : `roslaunch noether_examples mesh_filtering_demo.launch`

It says the process has died! May I know what is going wrong? 

```
started roslaunch server http://localhost:33195/

SUMMARY
========

PARAMETERS
 * /mesh_filter_server/mesh_filter_manager/filter_groups: [{'continue_on_fa...
 * /mesh_filtering_client/filter_group: DEFAULT
 * /mesh_filtering_client/mesh_file: /home/qbot/toolpa...
 * /mesh_filtering_client/results_dir: /home/qbot/toolpa...
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    mesh_filter_server (noether/mesh_filter_server)
    mesh_filtering_client (noether_examples/mesh_filtering_client)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[mesh_filtering_client-1]: started with pid [15856]
process[rviz-2]: started with pid [15857]
process[mesh_filter_server-3]: started with pid [15858]
Info:    Available plugins for base class 'noether_filtering::mesh::MeshFilterBase' :
		noether_filtering/BSplineReconstruction
		noether_filtering/CleanData
		noether_filtering/EuclideanClustering
		noether_filtering/FillHoles
		noether_filtering/WindowedSincSmoothing
Info:    noether_filtering::FilterManager<pcl::PolygonMesh> loaded filter group 'DEFAULT'
[ INFO] [1600188535.393935342]: Mesh Filter Server loaded parameters
[ INFO] [1600188535.398359996]: Mesh Filter Server is ready ...
Info:    Available plugins for base class 'noether_filtering::mesh::MeshFilterBase' :
		noether_filtering/BSplineReconstruction
		noether_filtering/CleanData
		noether_filtering/EuclideanClustering
		noether_filtering/FillHoles
		noether_filtering/WindowedSincSmoothing
Info:    noether_filtering::FilterManager<pcl::PolygonMesh> loaded filter group 'DEFAULT'
[ INFO] [1600188535.664401032]: Mesh Filter Server loaded parameters
Warning: noether_filtering::FilterGroup<pcl::PolygonMesh> received empty list of filters, using all filters loaded
         at line 161 in /home/qbot/toolpath/src/noether/noether_filtering/include/noether_filtering/impl/filter_group.hpp
Info:    Removed duplicate points, retained 29768 points from 29768
[mesh_filter_server-3] process has died [pid 15858, exit code -11, cmd /home/qbot/toolpath/devel/lib/noether/mesh_filter_server __name:=mesh_filter_server __log:=/home/qbot/.ros/log/800b43bc-f759-11ea-8898-049226d2aab3/mesh_filter_server-3.log].
log file: /home/qbot/.ros/log/800b43bc-f759-11ea-8898-049226d2aab3/mesh_filter_server-3*.log
```

jrgnicho, 2020-09-17:
Can you run the command below from the workspace and report the output?
```
ldd devel/lib/noether/mesh_filter_server | grep vtk
```

shubhamwagh, 2020-09-18:
@jrgnicho here is the output for the given command:
```
ldd devel/lib/noether/mesh_filter_server | grep vtk
	libvtkCommonCore-7.1.so.1 => /usr/local/lib/libvtkCommonCore-7.1.so.1 (0x00007f830c84d000)
	libvtkIOPLY-7.1.so.1 => /usr/local/lib/libvtkIOPLY-7.1.so.1 (0x00007f830b954000)
	libvtkIOGeometry-7.1.so.1 => /usr/local/lib/libvtkIOGeometry-7.1.so.1 (0x00007f830ac7a000)
	libvtkIOLegacy-7.1.so.1 => /usr/local/lib/libvtkIOLegacy-7.1.so.1 (0x00007f830a9ae000)
	libvtkIOImage-7.1.so.1 => /usr/local/lib/libvtkIOImage-7.1.so.1 (0x00007f830a5b6000)
	libvtkImagingCore-7.1.so.1 => /usr/local/lib/libvtkImagingCore-7.1.so.1 (0x00007f830a178000)
	libvtkIOCore-7.1.so.1 => /usr/local/lib/libvtkIOCore-7.1.so.1 (0x00007f8309eca000)
	libvtkCommonExecutionModel-7.1.so.1 => /usr/local/lib/libvtkCommonExecutionModel-7.1.so.1 (0x00007f8309bb5000)
	libvtkCommonDataModel-7.1.so.1 => /usr/local/lib/libvtkCommonDataModel-7.1.so.1 (0x00007f8309447000)
	libvtksys-7.1.so.1 => /usr/local/lib/libvtksys-7.1.so.1 (0x00007f8307be5000)
	libvtkCommonMisc-7.1.so.1 => /usr/local/lib/libvtkCommonMisc-7.1.so.1 (0x00007f830736b000)
	libvtkCommonTransforms-7.1.so.1 => /usr/local/lib/libvtkCommonTransforms-7.1.so.1 (0x00007f8306a97000)
	libvtkCommonSystem-7.1.so.1 => /usr/local/lib/libvtkCommonSystem-7.1.so.1 (0x00007f830687c000)
	libvtkzlib-7.1.so.1 => /usr/local/lib/libvtkzlib-7.1.so.1 (0x00007f830665f000)
	libvtkDICOMParser-7.1.so.1 => /usr/local/lib/libvtkDICOMParser-7.1.so.1 (0x00007f8306405000)
	libvtkmetaio-7.1.so.1 => /usr/local/lib/libvtkmetaio-7.1.so.1 (0x00007f83060de000)
	libvtkpng-7.1.so.1 => /usr/local/lib/libvtkpng-7.1.so.1 (0x00007f8305ea9000)
	libvtktiff-7.1.so.1 => /usr/local/lib/libvtktiff-7.1.so.1 (0x00007f8305c23000)
	libvtkCommonMath-7.1.so.1 => /usr/local/lib/libvtkCommonMath-7.1.so.1 (0x00007f83059f8000)
	libvtkjpeg-7.1.so.1 => /usr/local/lib/libvtkjpeg-7.1.so.1 (0x00007f83057c9000)

```

jrgnicho, 2020-09-18:
I haven't been able to reproduce your error
I'd suggest a couple of things
- Run ldd and see if it's linked multiple versions of pcl
```
ldd devel/lib/noether/mesh_filter_server | grep pcl
```
If only one version shows, most like pcl 1.9.1, then that means it's only linked one version of the pcl library which is a good sign

At this point I would build the noether_filtering package in "Release with debug info" or in "debug" and
then attach a [gdb process](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) to the `mesh_filter_server` node in the **noether/launch/mesh_filter_server.launch** to see where it fails.

shubhamwagh, 2020-09-21:
Hi,

Here is the output to `ldd devel/lib/noether/mesh_filter_server | grep pcl`

```
	libpcl_common.so.1.9 => /usr/local/lib/libpcl_common.so.1.9 (0x00007f399b059000)
	libpcl_kdtree.so.1.9 => /usr/local/lib/libpcl_kdtree.so.1.9 (0x00007f399ad1d000)
	libpcl_octree.so.1.9 => /usr/local/lib/libpcl_octree.so.1.9 (0x00007f399a956000)
	libpcl_search.so.1.9 => /usr/local/lib/libpcl_search.so.1.9 (0x00007f399a576000)
	libpcl_sample_consensus.so.1.9 => /usr/local/lib/libpcl_sample_consensus.so.1.9 (0x00007f399979d000)
	libpcl_filters.so.1.9 => /usr/local/lib/libpcl_filters.so.1.9 (0x00007f3998e35000)
	libpcl_io.so.1.9 => /usr/local/lib/libpcl_io.so.1.9 (0x00007f39989de000)
	libpcl_features.so.1.9 => /usr/local/lib/libpcl_features.so.1.9 (0x00007f3996471000)
	libpcl_ml.so.1.9 => /usr/local/lib/libpcl_ml.so.1.9 (0x00007f3996251000)
	libpcl_segmentation.so.1.9 => /usr/local/lib/libpcl_segmentation.so.1.9 (0x00007f399538e000)
	libpcl_visualization.so.1.9 => /usr/local/lib/libpcl_visualization.so.1.9 (0x00007f3995054000)
	libpcl_surface.so.1.9 => /usr/local/lib/libpcl_surface.so.1.9 (0x00007f3993db3000)
	libpcl_registration.so.1.9 => /usr/local/lib/libpcl_registration.so.1.9 (0x00007f3993a0d000)
	libpcl_keypoints.so.1.9 => /usr/local/lib/libpcl_keypoints.so.1.9 (0x00007f3993653000)
	libpcl_tracking.so.1.9 => /usr/local/lib/libpcl_tracking.so.1.9 (0x00007f399305f000)
	libpcl_recognition.so.1.9 => /usr/local/lib/libpcl_recognition.so.1.9 (0x00007f399296c000)
	libpcl_stereo.so.1.9 => /usr/local/lib/libpcl_stereo.so.1.9 (0x00007f399272d000)
	libpcl_outofcore.so.1.9 => /usr/local/lib/libpcl_outofcore.so.1.9 (0x00007f3992513000)
	libpcl_people.so.1.9 => /usr/local/lib/libpcl_people.so.1.9 (0x00007f39922f6000)
	libpcl_io_ply.so.1.9 => /usr/local/lib/libpcl_io_ply.so.1.9 (0x00007f3975a39000)
```

which looks fine to me.

jrgnicho, 2020-09-21:
Yes, it looks ok.  I would run gdb as described in my earlier comment

Beshario, 2020-09-23:
running 
``` cat /home/qbot/.ros/log/800b43bc-f759-11ea-8898-049226d2aab3/mesh_filter_server-3.log ```

can shed some light on what is happening.

shubhamwagh, 2020-09-24:
Hi @jrgnicho I rebuilt it with `RELWITHDEBINFO`, I still get the same error. Kindly find the attached gdb debugger output.

```

Info:    Available plugins for base class 'noether_filtering::mesh::MeshFilterBase' :
		noether_filtering/BSplineReconstruction
		noether_filtering/CleanData
		noether_filtering/EuclideanClustering
		noether_filtering/FillHoles
		noether_filtering/WindowedSincSmoothing
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/qbot/saws-robot-software/ros_packages/devel/lib/noether_examples/mesh_filtering_client...done.
Info:    noether_filtering::FilterManager<pcl::PolygonMesh> loaded filter group 'DEFAULT'
[ INFO] [1600939495.869257109]: Mesh Filter Server loaded parameters
[ INFO] [1600939495.873436754]: Mesh Filter Server is ready ...
Starting program: /home/qbot/saws-robot-software/ros_packages/devel/lib/noether_examples/mesh_filtering_client __name:=mesh_filtering_client __log:=/home/qbot/.ros/log/ce96818c-fe47-11ea-85e1-049226d2aab3/mesh_filtering_client-2.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffea3e2700 (LWP 24359)]
[New Thread 0x7fffe9be1700 (LWP 24360)]
[New Thread 0x7fffe93e0700 (LWP 24361)]
[New Thread 0x7fffe8bdf700 (LWP 24362)]
[New Thread 0x7fffdbfff700 (LWP 24363)]
[New Thread 0x7fffdb7fe700 (LWP 24364)]
[New Thread 0x7fffdaffd700 (LWP 24365)]
[New Thread 0x7fffda7fc700 (LWP 24366)]
Info:    Available plugins for base class 'noether_filtering::mesh::MeshFilterBase' :
		noether_filtering/BSplineReconstruction
		noether_filtering/CleanData
		noether_filtering/EuclideanClustering
		noether_filtering/FillHoles
		noether_filtering/WindowedSincSmoothing
Info:    noether_filtering::FilterManager<pcl::PolygonMesh> loaded filter group 'DEFAULT'
[ INFO] [1600939498.737997224]: Mesh Filter Server loaded parameters
Warning: noether_filtering::FilterGroup<pcl::PolygonMesh> received empty list of filters, using all filters loaded
         at line 161 in /home/qbot/saws-robot-software/ros_packages/src/noether/noether_filtering/include/noether_filtering/impl/filter_group.hpp
Info:    Removed duplicate points, retained 29768 points from 29768
[New Thread 0x7fffd9ffb700 (LWP 24368)]
[mesh_filter_server-4] process has died [pid 24314, exit code -11, cmd /home/qbot/saws-robot-software/ros_packages/devel/lib/noether/mesh_filter_server __name:=mesh_filter_server __log:=/home/qbot/.ros/log/ce96818c-fe47-11ea-85e1-049226d2aab3/mesh_filter_server-4.log].
log file: /home/qbot/.ros/log/ce96818c-fe47-11ea-85e1-049226d2aab3/mesh_filter_server-4*.log
```

Also @Beshario, I tried doing that earlier, but for some reason, it says the file is not present. So for the above output, I tried the following command line:

`cat  /home/qbot/.ros/log/ce96818c-fe47-11ea-85e1-049226d2aab3/mesh_filter_server-4.log`

jrgnicho, 2020-09-28:
@shubhamwagh could you run a backtrace when the node crashes.  You should be able to do that with the `bt` command

Levi-Armstrong, 2020-10-12:
@shubhamwagh Were you able to resolve the issue?

shubhamwagh, 2020-10-14:
Sorry, I am no more using this library. Instead created my own custom toolpath generation software from point clouds to meet our use case. Thanks for the help anyways.

Levi-Armstrong, 2020-10-14:
Is your point cloud method for generating tool paths open sourced?

shubhamwagh, 2021-05-05:
@Levi-Armstrong, apologies for late reply. It is not open sourced as it is part of company project and in development stage.