ros-industrial/noether #110: How to get raster paths wrt global axis?

shubhamwagh, 2020-09-22:
May I know how can I get raster paths with respect to the global axis? The parameter in tool.yaml is not used (else statement is empty). 
Could you outline the steps needed to get the raster paths in global frame?

jrgnicho, 2020-09-30:
The raster waypoints will be generated in the frame of the mesh model.  You can set the raster_angle to a non-zero value to apply a rotation relative to the z vector of the mesh model.  If you need to get the rasters relative to another frame  then you'd have to manually apply the desired transform to each waypoint as a post processing step.

marip8, 2020-12-11:
@shubhamwagh #121 should solve this issue (at least with the surface walk tool path planner). If you provide the `cut_direction` parameter as the vector (in mesh -- i.e. global -- coordinates) that represents the desired raster line direction, it should generate the raster path appropriately