ros-industrial/noether #127: Workspace Build Error with cmake_common_scripts

acbuynak, 2021-02-16:
Relatively new user here working in Melodic. I'm trying to build Noether and running into some build errors.
I've noted Issue/PR # 93 # 98

I'd any appreciate any guidance others have. Still learning. Thanks!

_VTK_viewer call for Cmake_Common_Scripts_
Normally if I'd see this error, I would look for a cmakelist file or similar.
```
CMake Error at /home/me/ws_noether/src/noether/vtk_viewer/CMakeLists.txt:4 (find_package):
  By not providing "Findcmake_common_scripts.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cmake_common_scripts", but CMake did not find one.

  Could not find a package configuration file provided by
  "cmake_common_scripts" with any of the following names:

    cmake_common_scriptsConfig.cmake
    cmake_common_scripts-config.cmake

  Add the installation prefix of "cmake_common_scripts" to CMAKE_PREFIX_PATH
  or set "cmake_common_scripts_DIR" to a directory containing one of the
  above files.  If "cmake_common_scripts" provides a separate development
  package or SDK, be sure it has been installed.
```

_PCL Build_
The only exception I made when following the build guide was prior to building PCL, I was unable to find a flag `PCL_ENABLE_SSE`. A number of warnings (? my word, not explicit) flagged in `opennurbs` when building, but no errors stopped the build.

I do not know how to pass the `-DNURBS_FOUND=ON/OFF` flag noted in [Issue# 98](https://github.com/ros-industrial/noether/issues/98#issuecomment-670563024). I don't think this related to my current issue.

_Side Note_
I do see a number of developer warnings when building the `noether_filtering` package, but it does successfully build (with a note that PCL NURBS was found).

marip8, 2021-02-16:
The name of `cmake_common_scripts` was recently changed (to `ros_industrial_cmake_boilerplate`), so that is likely your issue. The easiest short-term solution would be to check out [tag 0.2.5](https://github.com/ros-industrial/ros_industrial_cmake_boilerplate/tree/0.2.5) (the last state at which the package used to be called `cmake_common_scripts`) and try building again.

The CI builds are also failing because the current `.rosinstall` file pulls a named branch instead of a version or hash of that repository. We should update this repository to use the latest version of `ros_industrial_cmake_boilerplate` and remove it from the `.rosinstall` file when its debian is available (should be in a few weeks)

marip8, 2021-02-16:
> PCL Build
> The only exception I made when following the build guide was prior to building PCL, I was unable to find a flag PCL_ENABLE_SSE. A number of warnings (? my word, not explicit) flagged in opennurbs when building, but no errors stopped the build.

I'm not sure if this flag shows up if you use the CMake GUI, but it definitely exists and is probably worth turning on. You can do this when running `cmake`:
```
cmake .. -DPCL_ENABLE_SSE=OFF
```
You can also `grep` the `CMakeCache.txt` that gets generated by this process to check what the value of that argument is to make sure it was set correctly before you call `make`

> I do not know how to pass the -DNURBS_FOUND=ON/OFF flag noted in Issue# 98. I don't think this related to my current issue.

This is an argument for `noether_filtering` and you can set it in a similar way when you use `catkin` to build the package:
```
catkin build noether_filtering --cmake-args -DNURBS_FOUND=ON
```

acbuynak, 2021-02-16:
Awesome. Thanks for the breakdown in such clarity. This was helpful.

I successfully built the package in my workspace.
Additionally, I was able to rebuild `pcl-1.9.1` with both parameters set as confirmed by `CMakeCache.txt`

I'm better understanding how make/cmake/ccmake function together to build source code.

acbuynak, 2021-02-23:
Note:  The updated name reference is part of a PR here: https://github.com/ros-industrial/noether/pull/126

closed 2021-02-23 by acbuynak