ros-industrial/noether #134: Need for noether_conversions

jdlangs, 2021-05-18:
Since we're doing some bigger thinking about the future of this repository, I thought I'd propose merging `noether_conversions` into `noether`. It's only a single header and source file that contain a handful of utility functions and I'm not sure there's too many use cases where someone only wants that functionality and nothing else.

Pros: simplified repo structure, faster build time
Cons: API breakage

@Levi-Armstrong @marip8 @jrgnicho thoughts?

marip8, 2021-06-02:
I that's fine by me. Per some of our previous discussions, I believe we wanted to consolidate the ROS-related code anyway.

jrgnicho, 2021-06-02:
At the moment there isn't much in the noether_conversion package but there currently are various conversion utility functions scattered through the entire repo and so I would envision that a future version of noether would consolidate most of these into the noether_conversion package.  Because of that I'd prefer to keep it as a separate package which would eventually grow to contains more stuff

Levi-Armstrong, 2021-06-02:
I am in favor of merging noether_conversions into the Noether package seeing as it includes the ROS interfaces for the various noether libraries. I see no value in keeping it as a separate package since it primarily used by the Noether package for converting between types. It can be as a separate library within the noether package.