ros-industrial/noether #137: General question - Program a robot using CAD model to move accurately around a part

JTray123, 2021-06-03:
Hi guys, 

It was mentioned on the ROS forum that this framework might be a good starting place for a project I am working on. 

Basically I I need to program a robot tool path to move accurately around the full CAD part so I can scan the part in full,  not just the edges, with a sensor mounted to the robot. It looks like noether does this

Ideally I would like to 

- Import a CAD program (any of these formats will do obj, stl, ply, .step)
- Dynamically create a robot program from the CAD where the robot moves around the part and the robot tool continuously points to the surface normal of the part as the robot moves. 
- Have the robot tool offset from the physical part by 10mm at all locations
- A relatively simple way to create the robot path and make sure it is smooth as it moves continuously around all areas of the part. (The parts are complex in shape - like the famous bunny!)
- set the resolution so that the distance between any points on the path taken is consistent (for example 100um gap) - basically I need to scan the full surface area of the part in as high a resolution as possible. Assume time to scan the part < 5 mins

I have compiled noether and I have the demo programs working.

So my questions initially are before I go too far. 

1. Is this the right place to be asking such questions?
2. Is the Surface Raster Planner demo where I should be starting with to solve this problem.  When I run this demo it tells me that it has found 1 raster containing 257 points - is that correct for the demo program?. There is no visualization to what it is doing or what I do next with these points. I presume these are the tool path points the robot will move over the CAD part?
2. If this is the correct demo I should start with how would I set it so that I could adjust how fine the tool path plan will so that the sensor can record more points as it moves around, basically giving a finer, more dense point cloud? (Would you call this the resolution of the tool path?)

Hope this makes sense - I'll edit the question if you need more information

Thanks in advance

JT

marip8, 2021-06-04:
I'm glad you're interested in using this repository. Some thoughts about your requirements:

> Import a CAD program (any of these formats will do obj, stl, ply, .step)

There is technically a difference between CAD models and surface meshes. CAD models (like .step, .igs, .x_t, etc.) are mathematical, parametric representations of geometry. Surface meshes (like .stl, .ply., .obj, etc.) essentially only contain vertex and triangle information. This repository currently only works with surface meshes, not with CAD model formats

> Dynamically create a robot program from the CAD where the robot moves around the part and the robot tool continuously points to the surface normal of the part as the robot moves.

> A relatively simple way to create the robot path and make sure it is smooth as it moves continuously around all areas of the part. (The parts are complex in shape - like the famous bunny!)

This repository will generate a sequence of 6-DOF Cartesian tool waypoints which we typically call a tool path. This is technically not a robot program in the sense of execution on real hardware. If you want an executable robot program, you would have to use a post-processor library like RoboDK to turn these waypoints into robot move commands, or you would need to use a ROS motion planner like Tesseract or MoveIt to create robot trajectories from the tool path. The scope of this repository is limited to the generation of tool paths on input surfaces.

> Have the robot tool offset from the physical part by 10mm at all locations

I believe all of the planners generate tool waypoints directly on the surface of the part. If you want an offset like this, you would need to write code to apply your desired transform offset after the tool path is generated. 

> set the resolution so that the distance between any points on the path taken is consistent (for example 100um gap) - basically I need to scan the full surface area of the part in as high a resolution as possible. Assume time to scan the part < 5 mins

The waypoints of a tool path can be generated arbitrarily close to one another, so this is definitely possible. This repository doesn't consider timing of your process as it only generates the tool waypoints

# Q&A
> Is this the right place to be asking such questions?

Sure. I'm going to close this issue since technically it's not an issue, but we can certainly continue the conversation here. You can also email me or the other maintainers directly for more detailed information

> Is the Surface Raster Planner demo where I should be starting with to solve this problem. 

It really depends on what type of tool path you're looking for. Currently this repository has planner for generating raster patterns for surface coverage (plane slice and surface walk), and edge paths (half-edge and Eigen value edge path planners). If by "Surface Raster Planner demo" you mean a raster pattern surface coverage tool path, then [this example](https://github.com/ros-industrial/noether/blob/30f41bdf52ada8176331d06a628eec0a721730bf/noether_examples/launch/plane_slicer_rastering_generator_demo.launch), is probably what you're looking for

> There is no visualization to what it is doing or what I do next with these points. I presume these are the tool path points the robot will move over the CAD part?

Technically there is (at least for the plane slice example), but it isn't launched as a part of that demo. If you run Rviz along-side the demo and add two MarkerArray displays you should see this:

[image]

We probably should update the demo to launch Rviz with these displays already enabled

> When I run this demo it tells me that it has found 1 raster containing 257 points - is that correct for the demo program?. 

Yes and no. It seems like the tool waypoints seemed to be generated correctly, but there appear to be 9-10 rasters. My guess is that the demo node is probably squashing them all into a single "raster" and reporting the total number of points

> how would I set it so that I could adjust how fine the tool path plan will so that the sensor can record more points as it moves around, basically giving a finer, more dense point cloud? (Would you call this the resolution of the tool path?)

The plane slice and surface walk planners both have parameters for how close the lines and points are space (`line_spacing`, `point_spacing`). The half-edge path planner also has [an enum parameter](https://github.com/ros-industrial/noether/blob/30f41bdf52ada8176331d06a628eec0a721730bf/tool_path_planner/include/tool_path_planner/halfedge_edge_generator.h#L40-L46) for how closely the points are generated. If you're using the demo, I these parameters should be exposed by ROS parameter

closed 2021-06-04 by marip8

gavanderhoorn, 2021-06-04:
> > Is this the right place to be asking such questions?
> 
> Sure. I'm going to close this issue since technically it's not an issue, but we can certainly continue the conversation here. You can also email me or the other maintainers directly for more detailed information

the repository has a [Discussions](https://github.com/ros-industrial/noether/discussions) tab.

To allow others to also learn from the questions posted by @JTray123 and the answers by @marip8 et al., it would be great if we could keep Q&A on the discussions forum.

marip8, 2021-06-04:
Forgot about the discussions page. I'll migrate this conversation there