ros-industrial/noether #140: Unable to run examples

aa-tom, 2021-07-07:
Hi,

I'm trying to run the examples provided with the launch but am getting an error which _I think_ is due to missing VTK files. I compiled and installed both VTK and PCL using the guide provided in the readme so I expect all of the right files should exist. Any ideas on what might be causing the issue?

```sh
rob@aa-006:~/catkin_ws$ roslaunch noether_examples plane_slicer_rastering_generator_demo.launch
... logging to /home/rob/.ros/log/45b16e1c-df2a-11eb-adda-f5867854b9db/roslaunch-aa-006-108618.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://aa-006:37317/

SUMMARY
========

PARAMETERS
 * /plane_slicer_generator_node/mesh_file: /home/rob/catkin_...
 * /plane_slicer_generator_node/plane_slicer_rastering/min_hole_size: 0.02
 * /plane_slicer_generator_node/plane_slicer_rastering/min_segment_size: 0.01
 * /plane_slicer_generator_node/plane_slicer_rastering/point_spacing: 0.02
 * /plane_slicer_generator_node/plane_slicer_rastering/raster_rot_offset: 0.0
 * /plane_slicer_generator_node/plane_slicer_rastering/raster_spacing: 0.05
 * /plane_slicer_generator_node/plane_slicer_rastering/search_radius: 0.02
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    plane_slicer_generator_node (noether_examples/plane_slicer_example_node)
    tool_path_planner_server (noether/tool_path_planner_server)

auto-starting new master
process[master]: started with pid [108626]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 45b16e1c-df2a-11eb-adda-f5867854b9db
process[rosout-1]: started with pid [108636]
started core service [/rosout]
process[plane_slicer_generator_node-2]: started with pid [108643]
/home/rob/catkin_ws/devel/lib/noether_examples/plane_slicer_example_node: error while loading shared libraries: libvtksys-7.1.so.1: cannot open shared object file: No such file or directory
process[tool_path_planner_server-3]: started with pid [108644]
/home/rob/catkin_ws/devel/lib/noether/tool_path_planner_server: error while loading shared libraries: libvtksys-7.1.so.1: cannot open shared object file: No such file or directory
[plane_slicer_generator_node-2] process has died [pid 108643, exit code 127, cmd /home/rob/catkin_ws/devel/lib/noether_examples/plane_slicer_example_node __name:=plane_slicer_generator_node __log:=/home/rob/.ros/log/45b16e1c-df2a-11eb-adda-f5867854b9db/plane_slicer_generator_node-2.log].
log file: /home/rob/.ros/log/45b16e1c-df2a-11eb-adda-f5867854b9db/plane_slicer_generator_node-2*.log
[tool_path_planner_server-3] process has died [pid 108644, exit code 127, cmd /home/rob/catkin_ws/devel/lib/noether/tool_path_planner_server __name:=tool_path_planner_server __log:=/home/rob/.ros/log/45b16e1c-df2a-11eb-adda-f5867854b9db/tool_path_planner_server-3.log].
log file: /home/rob/.ros/log/45b16e1c-df2a-11eb-adda-f5867854b9db/tool_path_planner_server-3*.log

```

Any help is much appreciated!

aa-tom, 2021-07-07:
Resolved. I just needed to run ldconfig to make the shared libs available :smiley:

marip8, 2021-07-07:
Glad to hear you figured it out. For reference, if you're on Ubuntu 20.04 you can use the distributed debian versions of PCL and VTK rather than building them from source. For Ubuntu 18.04 you can also install VTK7 via debian and PCL 1.9 from [this PPA](https://launchpad.net/~felix928/+archive/ubuntu/pcl-1.9). This will save you some time and hassle in the future.

closed 2021-07-07 by marip8

aa-tom, 2021-07-07:
Ah, that's excellent to hear. Hoping to install into a Docker image eventually so that'll save a lot of build time :+1:

marip8, 2021-07-07:
We also have [a docker image with the noether dependencies installed](https://hub.docker.com/r/rosindustrial/noether) that we use to run CI. You might take a look at that as well