ros-industrial/noether #144: Error when trying to install PCL1.9 from PPA

aa-tom, 2021-07-08:
Hi,

I'm trying to install the the PCL archive from this [PPA](https://launchpad.net/~felix928/+archive/ubuntu/pcl-1.9), but I'm getting the following error:

```sh
rob@aa-006:~$ sudo add-apt-repository ppa:felix928/pcl-1.9 -y
Hit:1 http://gb.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://gb.archive.ubuntu.com/ubuntu focal-updates InRelease              
Ign:3 http://ppa.launchpad.net/felix928/pcl-1.9/ubuntu focal InRelease         
Hit:4 http://gb.archive.ubuntu.com/ubuntu focal-backports InRelease            
Hit:5 http://ppa.launchpad.net/mattrose/terminator/ubuntu focal InRelease      
Get:6 https://download.docker.com/linux/ubuntu focal InRelease [52.1 kB]       
Err:7 http://ppa.launchpad.net/felix928/pcl-1.9/ubuntu focal Release           
  404  Not Found [IP: 91.189.95.85 80]
Hit:8 http://dl.google.com/linux/chrome/deb stable InRelease                   
Hit:9 http://repo.mysql.com/apt/ubuntu focal InRelease                         
Hit:10 http://security.ubuntu.com/ubuntu focal-security InRelease              
Hit:11 http://packages.ros.org/ros/ubuntu focal InRelease
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/felix928/pcl-1.9/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

Any help would be much appreciated

aa-tom, 2021-07-09:
I think it's because I'm using focal and the PPA only supports bionic. I suppose I'll have to build from source instead?

marip8, 2021-07-09:
> I think it's because I'm using focal and the PPA only supports bionic.

That might be the case. If you're on Focal, the distributed version of `libpcl-dev` (1.10) works fine with this repository. No need to build or install PCL 1.9 specifically.

I'm going to close this issue since it's related to the PPA and not to this repository specifically. Feel free to continue the discussion though

closed 2021-07-09 by marip8