ros-industrial/noether #145: CMake error: could not find ros_industrial_cmake_boilerplate-config.cmake

aa-tom, 2021-07-16:
Hello,

When trying to build noether, I'm getting the following error message with the noether_filtering package. I've installed `vtk7` and `libpcl-dev` through `apt-get` before running the build.

```
By not providing "Findros_industrial_cmake_boilerplate.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "ros_industrial_cmake_boilerplate", but
  CMake did not find one.

  Could not find a package configuration file provided by
  "ros_industrial_cmake_boilerplate" with any of the following names:

    ros_industrial_cmake_boilerplateConfig.cmake
    ros_industrial_cmake_boilerplate-config.cmake

  Add the installation prefix of "ros_industrial_cmake_boilerplate" to
  CMAKE_PREFIX_PATH or set "ros_industrial_cmake_boilerplate_DIR" to a
  directory containing one of the above files.  If
  "ros_industrial_cmake_boilerplate" provides a separate development package
  or SDK, be sure it has been installed.
```

Any advice would be much appreciated!

aa-tom, 2021-07-16:
Ah, so the issue was being caused by the vcs command in my build script not pulling the dependencies in. I'm now getting the same error with noether_msgs and pcl_msgs

```
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "pcl_msgs" with any
  of the following names:

    pcl_msgsConfig.cmake
    pcl_msgs-config.cmake

  Add the installation prefix of "pcl_msgs" to CMAKE_PREFIX_PATH or set
  "pcl_msgs_DIR" to a directory containing one of the above files.  If
  "pcl_msgs" provides a separate development package or SDK, be sure it has
  been installed.

```

aa-tom, 2021-07-16:
I've checked against a workspace where noether is compiling correctly (one where I build PCL from source). The difference seems to be that there's no pcl_msgs package at /opt/ros/noetic/share when installing from `libpcl-dev`. I'm wondering if there are additional steps that need to be taken to make the library available?

aa-tom, 2021-07-16:
Just checked the Dockerfile for noether, and I don't see any extra steps I'm missing :confused: 

```
RUN if [ "$ROSDISTRO" != "kinetic" ] && [ "$ROSDISTRO" != "melodic" ] ; then \
    apt-get update \
    && apt-get install -y --no-install-recommends \
    libpcl-dev \
    sudo && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/* ; fi
```

aa-tom, 2021-07-16:
I believe a workaround may be manually installing the `ros-noetic-pcl-msgs` package through `apt-get`

marip8, 2021-07-16:
> I believe a workaround may be manually installing the ros-noetic-pcl-msgs package through apt-get

Correct. This is actually the nominal method of installing that package, not necessarily a work-around. The `ros_industrial_cmake_boilerplate` package is also distributed on the ROS build farm and can be installed via `apt`:

```
sudo apt install ros-noetic-ros-industrial-cmake-boilerplate
```

Just FYI in case you weren't aware, many ROS-based repositories (including this one) track their build-from-source dependences in a `.rosinstall` file in the repository. You can use tools like [vcstool](http://wiki.ros.org/vcstool) and [wstool](http://wiki.ros.org/wstool) to automatically clone dependency repositories into your workspace so the project will build correctly. 

Similarly, each ROS package contains a `package.xml` file that specifies its dependencies. You can use the [rosdep](http://wiki.ros.org/rosdep) tool to install all depedency packages that are distributed on the ROS build farm (i.e. binaries of packages that you don't need to build from source).

The [build](https://github.com/ros-industrial/noether#build) instructions in this repository cover mostly cover these steps. We probably should update with the `rosdep` step too though

closed 2021-07-21 by aa-tom