ros-industrial/noether #148: Warnings appearing as errors when compiling noether

aa-tom, 2021-07-25:
Hi,

I'm attempting to compile noether without having to build VTK or PCL from source. I've downloaded the `libpcl-dev` and `vtk7` packages through `apt-get`, however I'm getting the following errors when building the `vtk_viewer` package:

```
local|rob@ros:~/checkout/external$ catkin build noether
----------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/rob/checkout/internal/devel:/home/rob/checkout/external/devel:/opt/ros/noetic
Workspace:                   /home/rob/checkout/external
----------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/rob/checkout/external/build
Devel Space:        [exists] /home/rob/checkout/external/devel
Install Space:      [unused] /home/rob/checkout/external/install
Log Space:         [missing] /home/rob/checkout/external/logs
Source Space:       [exists] /home/rob/checkout/external/src
DESTDIR:            [unused] None
----------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
----------------------------------------------------------------------------------------------------------------
[build] Found '60' packages in 0.0 seconds.                                                                                                                                                                                   
[build] Updating package table.                                                                                                                                                                                               
Starting  >>> catkin_tools_prebuild                                                                                                                                                                                           
Finished  <<< catkin_tools_prebuild                           [ 1.0 seconds ]                                                                                                                                                 
Starting  >>> gtest                                                                                                                                                                                                           
Starting  >>> noether_msgs                                                                                                                                                                                                    
Starting  >>> pcl_conversions                                                                                                                                                                                                 
Finished  <<< gtest                                           [ 0.8 seconds ]                                                                                                                                                 
Starting  >>> ros_industrial_cmake_boilerplate                                                                                                                                                                                
Finished  <<< ros_industrial_cmake_boilerplate                [ 0.6 seconds ]                                                                                                                                                 
Starting  >>> noether_filtering                                                                                                                                                                                               
Starting  >>> vtk_viewer                                                                                                                                                                                                      
______________________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << pcl_conversions:cmake /home/rob/checkout/external/logs/pcl_conversions/build.cmake.000.log                                                                                                                      
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
cd /home/rob/checkout/external/build/pcl_conversions; catkin build --get-env pcl_conversions | catkin env -si  /usr/bin/cmake /home/rob/checkout/external/src/perception_pcl/pcl_conversions --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/rob/checkout/external/devel/.private/pcl_conversions -DCMAKE_INSTALL_PREFIX=/home/rob/checkout/external/install; cd -
..............................................................................................................................................................................................................................
Finished  <<< pcl_conversions                                 [ 2.0 seconds ]                                                                                                                                                 
______________________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << noether_filtering:cmake /home/rob/checkout/external/logs/noether_filtering/build.cmake.000.log                                                                                                                  
PCL NURBS NOT FOUND
Building mesh plugins
cd /home/rob/checkout/external/build/noether_filtering; catkin build --get-env noether_filtering | catkin env -si  /usr/bin/cmake /home/rob/checkout/external/src/noether/noether_filtering --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/rob/checkout/external/devel; cd -
..............................................................................................................................................................................................................................
______________________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << vtk_viewer:cmake /home/rob/checkout/external/logs/vtk_viewer/build.cmake.000.log                                                                                                                                
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni will be disabled
[... 410 lines cut ...]
Suppressed 213967 warnings (213967 in non-user code).
Use -header-filter=.* to display errors from all non-system headers. Use -system-headers to display errors from system headers as well.
66 warnings treated as errors
make[2]: *** [CMakeFiles/vtk_viewer.dir/build.make:76: CMakeFiles/vtk_viewer.dir/src/vtk_utils.cpp.o] Error 66
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                                       
[build] Runtime: 1 minute and 2.6 seconds total.                                                                                                                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

Any ideas & suggestions would be much appreciated!

Levi-Armstrong, 2021-07-26:
It is running clang tidy, but that should only run if NOETHER_ENABLE_TESTING is set to on. Though it does not look like you are setting it, but I do not see where it is being initialize to off. When building pass --cmake-args -DNOETHER_ENABLE_TESTING=OFF and see if it goes away. I also recommend doing a clean build.

aa-tom, 2021-07-26:
Ah okay, I'll try with that variable set then. Should those be fixed in noether then if the code isn't passing the clang check, or are they not a concern?

aa-tom, 2021-07-26:
Success! Thanks for the help. Will leave this open in case the warnings need to be fixed

Levi-Armstrong, 2021-07-26:
> Should those be fixed in noether then if the code isn't passing the clang check, or are they not a concern?

They should be fixed, but depending on the error/warning it may not be an issue.

Levi-Armstrong, 2021-07-26:
There are a few open PR's where we are restructuring and cleaning up the repo, so it may be getting addressed in one of those but this should remain open until those are merged and tested.