ros-industrial/noether #149: Issues generating paths for a subset of mesh faces

aa-tom, 2021-07-28:
Hi,

I'm currently trying to use noether to generate paths over a subset of faces on a mesh. The faces are chosen through a tool I've developed in rviz, which then extracts those faces from the Mesh (represented as a vtkPolyData object) and converts them into a shape_msgs/Mesh message. this mesh selection is eventually passed to noether's plane slicer and edge path planners. This is the code I'm using to generate the message.

```
shape_msgs::Mesh Mesh::toShapeMsgsMesh(std::vector<int> faces)
{
    std::vector<int> processed_points;
    shape_msgs::Mesh mesh;
    for (int i : faces)
    {
        shape_msgs::MeshTriangle triangle;
        vtkIdList* face = vtkIdList::New();
        mesh_->GetCellPoints(i, face);
        for (int j = 0; j < 3; j++)
        {
            int vertex_id = face->GetId(j);
            if (std::find(processed_points.begin(), processed_points.end(), vertex_id) != processed_points.end())
                continue;
            double* point = mesh_->GetPoint(vertex_id);
            geometry_msgs::Point vertex;
            vertex.x = point[0];
            vertex.y = point[1];
            vertex.z = point[2];
            mesh.vertices.push_back(vertex);
            processed_points.push_back(vertex_id);
        }
        triangle.vertex_indices = {face->GetId(0), face->GetId(1), face->GetId(2)};
        mesh.triangles.push_back(triangle);
    }
    return mesh;
}
```

`mesh_` is the vtkPolyData object containing all faces, and `faces` is the vector of selected face indexes.

When I attempt to generate a plane slice path over a selection, the message is passed to noether which then crashes. I don't _think_ it's an issue with the message building or noether calling code as, when I select every face on the mesh, the planning works fine. It's just when using a subset of faces that it breaks.

Any help would be much appreciated!

marip8, 2021-07-28:
At face value, it seems like the vertex indices of your triangles may not be correct. Those indices need to correspond to the index of the vertex in `mesh.vertices` not to the ID that VTK decided to assign the vertex

If that isn't the case, you'll need to provide more details about the crash you encountered; there's not much we can say about the root cause with the information in the post. I would suggest attaching a debugger to the process and finding the specific line at which the crash occurs and what the state of the variables are there

marip8, 2021-07-28:
What is the value of the [`search_radius` parameter](https://github.com/ros-industrial/noether/blob/c1719c1c009023b0bd68cf6158db02ae984871ff/tool_path_planner/include/tool_path_planner/plane_slicer_raster_generator.h#L56) that you're using? The default value is 0.01 m which is much smaller than 10 and 1 "units". For reference, mesh units are always interpreted in meters in this repository because the supported mesh files do not encode unit specification (as far as I know)

aa-tom, 2021-07-28:
Hey and thanks for the response. Search radius was set to 2 for the plane slice planner. Does the half edge generator also use one?

I've exported and uploaded the selection here too: https://filebin.net/cq69pc3dyuueyzcw

aa-tom, 2021-07-28:
Will check the mesh for duplicate vertices. Don't think there should be any. This is the updated code generating the mesh message:

aa-tom, 2021-07-28:
As in the above issue, I've verified that the selection does not contain duplicate vertices or faces using the following code

```
shape_msgs::Mesh Mesh::toShapeMsgsMesh(std::vector<int> faces)
{
    std::map<int, int> vertex_id_mapping;
    shape_msgs::Mesh mesh;
    for (int i : faces)
    {
        shape_msgs::MeshTriangle triangle;
        vtkIdList* face = vtkIdList::New();
        mesh_->GetCellPoints(i, face);
        for (int j = 0; j < 3; j++)
        {
            int vertex_id = face->GetId(j);
            if (vertex_id_mapping.find(vertex_id) != vertex_id_mapping.end())
            {
                triangle.vertex_indices[j] = vertex_id_mapping[vertex_id];
                continue;
            }
            double* point = mesh_->GetPoint(vertex_id);
            geometry_msgs::Point vertex;
            vertex.x = point[0];
            vertex.y = point[1];
            vertex.z = point[2];
            mesh.vertices.push_back(vertex);
            vertex_id_mapping[vertex_id] = mesh.vertices.size() - 1;
            triangle.vertex_indices[j] = vertex_id_mapping[vertex_id];
        }
        mesh.triangles.push_back(triangle);
    }
    for (int i = 0; i < mesh.vertices.size(); i++)
    {
        for (int j = 0; j < mesh.vertices.size(); j++)
        {
            if (i == j)
                continue;
            geometry_msgs::Point p1 = mesh.vertices[i];
            geometry_msgs::Point p2 = mesh.vertices[j];
            if (p1.x == p2.x && p1.y == p2.y && p1.z == p2.z)
            {
                ROS_WARN("Mesh contains duplicate vertex!");
                return mesh;
            }
        }
    }
    ROS_INFO("No duplicate vertices");
    for (int i = 0; i < mesh.triangles.size(); i++)
    {
        for (int j = 0; j < mesh.triangles.size(); j++)
        {
            if (i == j)
                continue;
            auto f1 = mesh.triangles[i].vertex_indices;
            auto f2 = mesh.triangles[j].vertex_indices;
            if (f1[0] == f2[0] && f1[1] == f2[1] && f1[2] == f2[2])
            {
                ROS_WARN("Mesh contains duplicate face!");
                return mesh;
            }
        }
    }
    ROS_INFO("No duplicate faces");
    return mesh;
}
```

```
[ INFO] [1627486721.860609990] [/rviz_1627486714898213024]: Loading part from /home/rob/checkout/halfsphere.stl
[ INFO] [1627486721.893491121] [/rviz_1627486714898213024]: 8373 vertices found
[ INFO] [1627486721.893561973] [/rviz_1627486714898213024]: 16562 faces found
[ INFO] [1627486721.895515564] [/rviz_1627486714898213024]: Built cell locator
[ INFO] [1627486721.903951288] [/rviz_1627486714898213024]: Computed 16562 normals
[ INFO] [1627486739.676315302] [/rviz_1627486714898213024]: No duplicate vertices
[ INFO] [1627486739.687466058] [/rviz_1627486714898213024]: No duplicate faces
```