ros-industrial/noether #178: Point ordering issue with plane_slicer

InigoMoreno, 2023-02-16:
Hello. I am trying to use noether with complex geometry (such as a cylinder) and I am seeing some issues.

I am running the plane_slicer_rastering_generator_demo.launch with a cylindrical mesh ([download link](https://drive.google.com/file/d/1VDzhZMbIJCRWRONGJZCb-JwrWQZjDdo6/view?usp=sharing))  and getting the following in noether.

[image]

As you can see, while the points seem to be correct, they seem to be ordered by the x coordinate instead of going around the cyllinder. Can someone help me figure out where to look to try and solve this?

InigoMoreno, 2023-02-16:
I changed the parameters so that it only generates one raster to better show the issue.

marip8, 2023-02-16:
What is the behavior that you are expecting from this tool path planner? Are you wanting circular paths that traverse around the perimeter of the cylinder at different heights, or do you want the same type of coverage pattern shown above, just in a different orientation?

InigoMoreno, 2023-02-17:
I want circular paths that traverse around the perimeter. The piece is a cylinder, so trajectories shown above actually go through the inside of the piece.

marip8, 2023-02-17:
`noether` in general was not necessarily designed to handle closed or watertight shapes like this, so I'm not surprised you're seeing this error. Most of the planners are written with the assumption that you're working with a single open flat-ish surface. The reason the paths come out like this is due to the [`RasterOrganizationModifier`](https://github.com/ros-industrial/noether/blob/e0b5b17c08e545bff471d9878ba43b3bd25ae27b/noether_tpp/src/tool_path_modifiers/organization_modifiers.cpp#L65-L75) which assumes that each line in the tool path segment has a relatively straight line direction (as in a regular planar raster) and attempts to sort the points by their projected distance along that average direction. This is not the case for the cylinder where each tool path segment in the "raster" is actually a circle, hence the behavior you see.

To fix this, we would need to create a new [tool path modifier ](https://github.com/ros-industrial/noether/blob/e0b5b17c08e545bff471d9878ba43b3bd25ae27b/noether_tpp/include/noether_tpp/core/tool_path_modifier.h#L32) which sorts the points in a tool path segment in a circular fashion based on some criteria (e.g., choose points greedily by distance to previous point, etc.). Then you could apply this new modifier behind the plane slice planner in a [tool path planner pipeline](https://github.com/ros-industrial/noether/blob/e0b5b17c08e545bff471d9878ba43b3bd25ae27b/noether_tpp/include/noether_tpp/core/tool_path_planner_pipeline.h) to produce the results you are looking for

marip8, 2023-03-07:
Let me know if you need any more clarification or help implementing [my suggestion from above](https://github.com/ros-industrial/noether/issues/178#issuecomment-1434838958) if you end up creating a new tool path modifier

InigoMoreno, 2023-03-10:
Just commenting out the RasterOrganizationModifier seems to work fine for my use case.