ros-industrial/noether #182: Error:   Vertex in PolygonMesh needs to have 3 elements only

SyZbidi, 2023-04-24:
Hi,

I've been trying to generate a `tool_path` on a mesh file that was generated using [industrial_reconstruction](https://github.com/ros-industrial/industrial_reconstruction) package but I got the following warning + error

```
[ INFO] [1682331786.689697710]: Got mesh file /home/syrine/workspace/object_reconstruction_ws/src/reconstruction_ply/default.ply
[pcl::PLYReader] /home/syrine/workspace/object_reconstruction_ws/src/reconstruction_ply/default.ply:15: property 'list uint8 uint32 vertex_indices' of element 'face' is not handled
Failed to find match for field 'x'.
Failed to find match for field 'y'.
Failed to find match for field 'z'.
Error:   Vertex in PolygonMesh needs to have 3 elements only
         at line 77 in /home/syrine/workspace/object_reconstruction_ws/src/noether/noether_conversions/src/noether_conversions.cpp
[ERROR] [1682331786.691695834]: Failed to read file /home/syrine/workspace/object_reconstruction_ws/src/reconstruction_ply/default.ply
[plane_slicer_generator_node-2] process has died [pid 429536, exit code 255, cmd /home/syrine/workspace/object_reconstruction_ws/devel/lib/noether_examples/plane_slicer_example_node __name:=plane_slicer_generator_node __log:=/home/syrine/.ros/log/007bcf26-e28a-11ed-b95f-a1c452af2903/plane_slicer_generator_node-2.log].
log file: /home/syrine/.ros/log/007bcf26-e28a-11ed-b95f-a1c452af2903/plane_slicer_generator_node-2*.log
^X^C[rviz-4] killing on exit
[tool_path_planner_server-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I'm assuming that the error is related to the warning, and the reason behind the warning is that the `x,y,z` fields for my mesh are `double` type while most of `pcl` examples if not all are `float`.

If i'm correct, is there a way around this? Otherwise what could be the problem ? 

Here are attached two examples of the mesh files that I had generated 
[mesh_files.zip](https://github.com/ros-industrial/noether/files/11311831/mesh_files.zip)

Command I ran:

 `roslanch noether_example plane_slicer_rastering_generator_demo.launch`

thank you for your help!

LuciferMorningStar0786, 2023-08-31:
Hi, 
I am also facing same issue, I am trying this in ROS1 Neotic. I've been trying to generate a tool_path on a mesh file that was generated using [industrial_reconstruction](https://github.com/ros-industrial/industrial_reconstruction) package but I got the following warning + error

#setting /run_id to e702e520-47e7-11ee-9b00-bb27e3a9d02a
process[rosout-1]: started with pid [21114]
started core service [/rosout]
process[plane_slicer_generator_node-2]: started with pid [21121]
process[tool_path_planner_server-3]: started with pid [21122]
process[rviz-4]: started with pid [21123]
[ INFO] [1693477184.127820727]: Got mesh file /home/kaayatech/catkin_ws/src/noether/noether_examples/data/proper__mesh.ply
[pcl::PLYReader] /home/kaayatech/catkin_ws/src/noether/noether_examples/data/proper__mesh.ply:15: property 'list uint8 uint32 vertex_indices' of element 'face' is not handled
Failed to find match for field 'x'.
Failed to find match for field 'y'.
Failed to find match for field 'z'.
Error:   Vertex in PolygonMesh needs to have 3 elements only
         at line 77 in /home/kaayatech/catkin_ws/src/noether/noether_conversions/src/noether_conversions.cpp
[ERROR] [1693477184.166035809]: Failed to read file /home/kaayatech/catkin_ws/src/noether/noether_examples/data/proper__mesh.ply
[plane_slicer_generator_node-2] process has died [pid 21121, exit code 255, cmd /home/kaayatech/catkin_ws/devel/lib/noether_examples/plane_slicer_example_node __name:=plane_slicer_generator_node __log:=/home/kaayatech/.ros/log/e702e520-47e7-11ee-9b00-bb27e3a9d02a/plane_slicer_generator_node-2.log].
log file: /home/kaayatech/.ros/log/e702e520-47e7-11ee-9b00-bb27e3a9d02a/plane_slicer_generator_node-2*.log

Please reach out

SyZbidi, 2024-02-15:
Probably too late now, i solved my problem using [plyfile](https://github.com/dranjan/python-plyfile/discussions/72#discussioncomment-6583212) to convert the field types of `x`, `y` and `z`

closed 2024-02-15 by SyZbidi

gavanderhoorn, 2024-02-15:
Should this be closed? The approach mentioned by @SyZbidi feels more like a work around.