ros-industrial/noether #185: Inconsistent point spacing on mesh with holes

dpiet, 2023-05-10:
Running the plane slicer raster planner on a simple part with small holes, we're encountering inconsistent point spacing of waypoints near the holes. min. hole spacing is greater than these holes and so we do get a single raster segment here (point spacing set to 0.03):
[image]

 
If we rotate the principal axis direction, we don't see the issue:
[image]

When we fill the holes we get a expected behavior with consistent waypoint spacing as well.

This is the mesh faces views in the affected area:
[image]

Any suggestions would be welcome!

dpiet, 2023-06-01:
Hi there, any suggestions on the above issue?

marip8, 2023-06-30:
@dpiet I started looking into this today and found a potentially unrelated issue coming from [this line](https://github.com/ros-industrial/noether/blob/5d4bf0ad1f74efb9073248a9716b9d2516960c5a/noether_tpp/src/tool_path_planners/raster/plane_slicer_raster_planner.cpp#L543) when this planner is used with a perfectly flat mesh (which was my test case for this issue). Can you upload your mesh to this issue so I can test with it instead?

FYI, I opened #191 to help visualize tool paths and reproduce issues. You might find that useful going forward as well

dpiet, 2023-07-18:
@marip8 sorry for the delay, here is test mesh you can use to test. This is a perfectly flat mesh exported from a CAD surface, so maybe it's related to the issue you're seeing?
[test_mesh.zip](https://github.com/ros-industrial/noether/files/12074934/test_mesh.zip)

marip8, 2023-07-21:
I get fairly consistent results with plane slicer raster planner using a variety of different raster direction generators.

Fixed direction generator (+x)
[image]

Fixed direction generator (-x)
[image]

Principal axis generator (+90)
[image]

Principal axis generator (-90)
[image]

Angled
[image]

In general, I think the expected behavior is that when the planner encounters a hole, it puts a waypoint at the edge of the hole, regardless of whether the distance from the last waypoint to the edge of the hole is less than point spacing distance. Interestingly, I was not able to recreate the results of your second picture where the planner did not put a waypoint at the edge of the hole. It almost seems as if it didn't recognize the hole, or it thought that it was smaller than the minimum hole size

dpiet, 2023-07-24:
Thanks @marip8! Are some of these duplicate images accidentally? Our point spacing is much larger at 30 mm vs the 5 mm you used in this testing, so perhaps it all depends on when an edge point falls near a hole discontinuity. In terms of intended behavior, would it make sense to honor the desired point spacing though? So if a point occurs within `point_spacing` distance to the edge of a hole, no additional point is added.

marip8, 2023-07-24:
> Are some of these duplicate images accidentally?

I don't think so. The direction generator component varies for each image. The planner just makes (visibly) the same plan given slightly different inputs

> Our point spacing is much larger at 30 mm vs the 5 mm you used in this testing, so perhaps it all depends on when an edge point falls near a hole discontinuity. 

I can test to see for sure. I would expect there still to be a point at the edge of a hole even with a 30mm point spacing

>  In terms of intended behavior, would it make sense to honor the desired point spacing though? So if a point occurs within point_spacing distance to the edge of a hole, no additional point is added.

We could definitely consider making it an option. For our (SwRI) use-cases, we are mostly generating waypoints for surface coverage processes like sanding, painting, etc., where the tool needs to go all the way up to the edge of a surface, including holes. We don't care as much about point spacing consistency because we almost always interpolate the tool path for motion planning after the fact.

dpiet, 2023-07-26:
I think we'll just need to check our interpolation on the backend as well. The reasoning makes sense. Going to close issue.

closed 2023-07-26 by dpiet