ros-industrial/noether #189: Build error with Ubuntu 22.04

srsidd, 2023-06-03:
Hi! I'm using noether with tesseract. This is with ros-humble using colcon as the build tool on an Ubuntu 22.04 machine. For all the catkin specific package, I have temporarily created a COLCON_IGNORE file in them. However, when I build I run into a error while building - 
```
--- stderr: noether_filtering                                                                                                                                                          
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find MPI (missing: MPI_C_FOUND C)

      Reason given by package: MPI component 'C' was requested, but language C is not enabled.  

Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.22/Modules/FindMPI.cmake:1830 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:397 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  CMakeLists.txt:8 (find_package)

---
Failed   <<< noether_filtering [0.63s, exited with code 1]
```

Any thoughts on how to resolve this?

I also found a dependency issue with VTK, it seems like ros-humble ships with VTK 9, but rosdep resolves `libvtk` which is listed as dependency in package.xml of noether_tpp resolves to VTK 7. This causes a circular dependency error - 

```
The following packages have unmet dependencies:
 python3-vtk9 : Breaks: python3-vtk7 but 7.1.1+dfsg2-10.1build1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

Since the build error and the dependency error are from VTK, I was wondering if they're both related?

srsidd, 2023-06-19:
Hello any suggestions on the fixing this issue?

marip8, 2023-06-20:
I'm fairly certain that `noether_filtering` does not actually link directly against VTK. You could try removing the [`find_package` on VTK](https://github.com/ros-industrial/noether/blob/7ec7101783d9159da83a38e1ba6d535b42171c99/noether_filtering/CMakeLists.txt#L8-L15) and the [dependency export of VTK](https://github.com/ros-industrial/noether/blob/7ec7101783d9159da83a38e1ba6d535b42171c99/noether_filtering/CMakeLists.txt#L165) to see if that resolves this issue. I personally haven't updated to 22.04, so I haven't looked into any of these build issues yet.

marip8, 2023-06-20:
Coincidentally, I also noticed another issue with `noether_filtering` today that you might run into. [This line](https://github.com/ros-industrial/noether/blob/7ec7101783d9159da83a38e1ba6d535b42171c99/noether_filtering/CMakeLists.txt#L113) references the `mesh_filter_plugins` library when it should reference the `cloud_filters` library instead. And [here](https://github.com/ros-industrial/noether/blob/7ec7101783d9159da83a38e1ba6d535b42171c99/noether_filtering/CMakeLists.txt#L121) it also references the `mesh_filter_plugins` library when it should reference the `cloud_filter_plugins` library

srsidd, 2023-07-10:
Apologies for the delay in getting back. Unfortunately, VTK is referenced in a couple of different places in the repo ([`noether_tpp`](https://github.com/ros-industrial/noether/blob/master/noether_tpp/CMakeLists.txt#L12) [`vtk_viewer`](https://github.com/ros-industrial/noether/blob/master/vtk_viewer/CMakeLists.txt#L6) and the core [`noether`](https://github.com/ros-industrial/noether/blob/master/noether/CMakeLists.txt#L8) repo. 

I think the underlying MPI bug needs to be addressed. Unfortunately, google was not very helpful in trying to determine this issue, so any pointers on how to go about it will be helpful.

Thanks!

marrts, 2023-07-10:
I was able to get this branch building in a ros:humble docker: https://github.com/marrts/noether/tree/humble-devel

srsidd, 2023-07-10:
Just tried that branch and it worked great! Thanks, I will close this issue.

closed 2023-07-10 by srsidd