ros-industrial/noether #196: Ubuntu 22 compile issue with QVTKWidget

natalieCloud, 2023-08-23:
As title states, I'm attempting to compile Noether on `Ubuntu 22.04` for use with ROS `Humble`. I'm building from [my own fork](https://github.com/natalieCloud/noether/tree/humble-devel-arp) which is based on @marrts latest `humble-devel` branch. I don't think my issue is related to the changes from these branches.

Following #189, I resolved most of my issues such as the libxmlrpcpp dependency via:
`sudo apt-get install libxmlrpcpp-dev`

### Current Issue

When building, I'm missing a dependency to `<QVTKWidget.h>`. I have version's Qt `5.15.3` and VTK `9.1` installed.

I think this is a simple missing dependency, but haven't been able to find it. Any suggestions? Thanks!!

acbuynak, 2023-08-23:
I have no experience with VTK, but a quick search shows `QVTKWidget` was depreciated for VTK 8.1 and a new class, `QVTKOpenGLWidget` was created. Looking at the class reference, it might just be a name change? [Ref nightly build docs here.](https://vtk.org/doc/nightly/html/classQVTKOpenGLWidget.html)

natalieCloud, 2023-08-29:
From what I can tell so far that may be the case? 

By reverting some of the changes @marrts made to their humble devel branch I was able to compile Noether on `Ubuntu 20.04` with ROS `Foxy` (Pushed on my fork)! 

However, the changes I needed to make further lead me to believe there may be a versioning issue with the switch from VTK `7.1` to `9.1`. Unfortunately though, when I tried to implement fixes I ran into even more deprecation and versioning issues. 

Was wondering if anyone else had any experience with/insight on this! Thank you again!

daniel-dsouza, 2023-09-01:
Could you please try to rebase your changes onto @marrts 
[humble-devel branch](https://github.com/marrts/noether/tree/humble-devel
)? It builds for me in humble.

marrts, 2023-09-01:
> When building, I'm missing a dependency to `<QVTKWidget.h>`

I am running into this same issue when rebasing on the latest changes of noether. It should be fixed now in my latest [humble-devel ](https://github.com/marrts/noether/tree/humble-devel) branch,

natalieCloud, 2023-09-06:
That worked! Thank you so much!!!

marip8, 2024-08-08:
Closing; addressed in #211

closed 2024-08-08 by marip8