ros-industrial/noether #205: Issue finding pre-build noether_tpp

rr-tom-noble, 2023-09-27:
Hi,

I'm attempting to build noether separately from my main codebase using catkin, then install it with the rest of my ros packages (to `/opt/ros/noetic`), however I'm running into the following issue using `find_package(noether_tpp)`.

```
  By not providing "Findnoether_tpp.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "noether_tpp", but CMake did not find one.

  Could not find a package configuration file provided by "noether_tpp" with
  any of the following names:

    noether_tppConfig.cmake
    noether_tpp-config.cmake

  Add the installation prefix of "noether_tpp" to CMAKE_PREFIX_PATH or set
  "noether_tpp_DIR" to a directory containing one of the above files.  If
  "noether_tpp" provides a separate development package or SDK, be sure it
  has been installed.
```

Checking the `/opt/ros/noetic/share/noether_tpp` directory confirms that there are no `.cmake` files, unlike some of the other packages in this repository.

Building noether alongside my project seems to work, however.

I'm not super knowledgeable around the whole cmake / catkin build process, so I'd appreciate any help understanding what I'm doing wrong.

rr-tom-noble, 2023-09-27:
Resolved now. I was manually creating a `.deb` from the install space, but had missed adding the `lib/cmake` directory :sweat:

closed 2023-09-27 by rr-tom-noble

marip8, 2023-09-27:
FYI, there are [some tools in RICB](https://github.com/ros-industrial/ros_industrial_cmake_boilerplate/blob/ae129076951b74096e435d77bc4bf0c0fbbee2d5/cmake/cpack_tools.cmake) to using CPack to create debians. We can consider adding those to `noether_tpp` and `noether_gui` if you think they would be useful. We do this already for some other projects, like [`tesseract`](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_environment/CMakeLists.txt#L160-L199)

rr-tom-noble, 2023-10-18:
Hi @marip8

We currently have a decent number of dependencies which we build from source and package ourselves, so any work to make built packages available by default would reduce the complexity of our dependency management, and would be much appreciated :)

marip8, 2023-10-18:
I understand the pain. We could do some work to release `noether_tpp` and `noether_gui` to the ROS2 build farm if that would  help. The only issue is that the ROS2 build farm only supports one version at a time, so if we update either of those packages in a way that breaks API, it will break your builds. I think the API we've defined for both should be relatively stable, but we may find in the future that we need to change things in a way that is breaking.

rr-tom-noble, 2023-10-19:
That would be great. We're currently using noetic, but with a plan to migrate to humble in the near future. The versioning isn't so much of an issue for us, as we're currently using Docker images to fix our versions at the time of build. We're also considering self-hosting snapshots of the ROS buildfarm packages we use for more control over when updates happen, as we tend to rebuild our images quite frequently.

I'd be happy to contribute to the effort. Our current process involves building with catkin, building a .deb on top of the install space using `dpkg-deb --build`, then publishing to a private apt repo (configured to run automatically on a GitHub workflow). I think I've seen that the tool bloom can be used to automate a lot of the process when publishing to the ROS buildfarm? 

Any suggestions would be appreciated!