ros-industrial/noether #225: Error in dependencies needed to run Noether Gui App (Dev Branch)

Humvee3982, 2024-04-15:
Proceed to reproduce:
1. Setting up a new VM with ros 2 rolling
2. Pull from dev branch
3. Create a new Workspace and execute `sudo resdep install -i -r --rosdistro=rolling --from-paths src`
4. Compiled with `colcon build --symlink-install`
5. Tried to run noether_gui_app and got the following error:
```
./install/noether_gui/bin/noether_gui_app: error while loading shared libraries: libnoether_gui.so: cannot open shared project file: No such file or directory
```

Humvee3982, 2024-04-15:
Added the lib folder of libjavm.so to the LD_LIBRARY_PATH, now libjavm.so is missing.

marip8, 2024-05-01:
I noticed the same thing [here](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/issues/74) on Humble, but the issue doesn't appear to happen any more. I haven't tested on `rolling` yet though. We should track down what dependency brings in `java`. Hopefully we could eliminate it or find a better fix to this issue

liamcarlos, 2024-05-31:
@Humvee3982 How are you running noether_gui_app?  I can't find the executable when I try autocomplete by tab in terminal nor can I find anything adding that executable in the dev branch CMake..

Humvee3982, 2024-06-03:
> @Humvee3982 How are you running noether_gui_app? I can't find the executable when I try autocomplete by tab in terminal but I can see the line adding adding executable noether_gui_app in the CMake..

You should find an Application when using colcon and running it without an error in the `installation/noether_gui/bin/` folder.

liamcarlos, 2024-06-06:
@Humvee3982  I see, there isn't a `ros2 launch noether_gui ...` executable, but it can be launched with `./noether_gui_app` from `install/noether_gui/bin`  This kind of sucks cause I didn't see any nodes nor topics when using the gui..  I think the documentation should include a tutorial on how to produce a PoseArray topic from these packages, but I have the feeling I'm going to have to figure that out on my own...

Side note, there's a typo in your LD_LIBRARY_PATH update that might drive some people who are having the same problem a little crazy.  Yours say `adm64` not `amd64`  So that would be this instead:

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/jvm/java-11-openjdk-amd64/lib
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/jvm/java-11-openjdk-amd64/lib/server
```

Very minor detail that's a little hard to pickup, so I thought I'd mention it.

Thanks for the help though!  I really appreciate it!  Got everything running on here thanks to you

marip8, 2024-06-06:
> I see, there isn't a ros2 launch noether_gui ... executable, but it can be launched with ./noether_gui_app from install/noether_gui/bin This kind of sucks cause I didn't see any nodes nor topics [...]

FWIW, all of this repo on the `dev` branch is independent of ROS, hence no topics, etc. The plan is to introduce a thin ROS1/2 layer on top of this repo in a different repository, but we haven't gotten there yet. TBD what that ROS layer will include, but likely will have conversions to common message types and maybe putting the GUI into an Rviz panel. I hesitate to put too much into the ROS layers because the requirements tend to vary so much from application to application. Feel free to post some suggestions if there are specific ROS features you think would be useful.

liamcarlos, 2024-06-07:
@marip8 I think the basic use of Noether is rastering or combing along the surface of a provided mesh, which I think is handled pretty well at the moment.  It would be nice to have to the current gui pop up in a separate window than RViz (similar to the scan n plan workshop) but also have a portion that let's you name and publish topics after generating tool path plans.  That would make the gui act like a preview for generating tool path plans.

I do believe Noether should have a closer connection to Industrial Reconstruction, so that users would have the option to load a mesh from a reconstruction, but I get how that would be application specific.  I do think that there are going to be a handful popular uses of Noether (which are seen in other ros-industrial repos) that could be packaged more project agnostic rather than ROS agnostic.  It was a little difficult to swap out robots and controllers from other projects to just use their panels for generating meshes and tool path plans..

Regardless, I appreciate your response and interest!  Noether really pulls my project together.  I'm going to see what else I can do to integrate it better into my workflow, and if I can add anything worth contributing (hopefully) I'll be back in here

rr-tom-noble, 2025-02-11:
I've also been noticing this with various ROS packages whilst migrating to ROS2, most recently with this one. I did some deeper digging (although this was a while ago) and seemed to come to the conclusion that it was something to do with PCL, which pulls in java libraries for some of its modules. The issue seemed to be related to a couple of points:

- I think there's a more general issue in colcon where, even if a module uses `find_package(PCL COMPONENTS ...)` to pull in a subset of PCL modules, whenever a package is pulled in transitively, `COMPONENTS` gets ignored, which leads to a lot of packages pulling in the java dependent PCL stuff despite not requiring it

- There was something odd going on with the order of packages in `find_package` and `ament_export_dependencies` that was causing PCL to "lose sight" of some of the libs it should've been including. In one instance switching the order of finding and exporting VTK and PCL in one of our own packages fixed this issue