ros-industrial/noether #235: catkin_build error in ubuntu 20.04

euichan2000, 2024-05-27:
I'm attempting to build Noether on `ubuntu 20.04`.
Here is my sequence to build Noether.
```
mkdir -p ~/path_ws/src
cd ~/path_ws/src
git clone git@github.com:ros-industrial/noether.git
cd path_ws
vcs import src < src/noether/dependencies.repos
rosdep install --from-paths src -iry
catkin build
```

then, this error message appears.
```
----------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/nrs/catkin_ws/devel:/opt/ros/noetic
Workspace:                   /home/nrs/path_ws
----------------------------------------------------------------------
Build Space:        [exists] /home/nrs/path_ws/build
Devel Space:        [exists] /home/nrs/path_ws/devel
Install Space:      [unused] /home/nrs/path_ws/install
Log Space:         [missing] /home/nrs/path_ws/logs
Source Space:       [exists] /home/nrs/path_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
----------------------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------------------
[build] Found 4 packages in 0.0 seconds.                                                           
[build] Updating package table.                                                                    
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                
Finished  <<< catkin_tools_prebuild                           [ 1.4 seconds ]                      
Starting  >>> ros_industrial_cmake_boilerplate                                                     
Finished  <<< ros_industrial_cmake_boilerplate                [ 0.6 seconds ]                      
Starting  >>> boost_plugin_loader                                                                  
Starting  >>> noether_tpp                                                                          
Finished  <<< boost_plugin_loader                             [ 5.3 seconds ]                      
___________________________________________________________________________________________________
Errors     << noether_tpp:cmake /home/nrs/path_ws/logs/noether_tpp/build.cmake.000.log             
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Error at /home/nrs/path_ws/src/noether/noether_tpp/CMakeLists.txt:77 (target_sources):
  target_sources called with invalid arguments

cd /home/nrs/path_ws/build/noether_tpp; catkin build --get-env noether_tpp | catkin env -si  /usr/bin/cmake /home/nrs/path_ws/src/noether/noether_tpp --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/nrs/path_ws/devel; cd -

...................................................................................................
Failed     << noether_tpp:cmake                               [ Exited with code 1 ]               
Failed    <<< noether_tpp                                     [ 6.5 seconds ]                      
Abandoned <<< noether_gui                                     [ Unrelated job failed ]             
[build] Summary: 3 of 5 packages succeeded.                                                        
[build]   Ignored:   None.                                                                         
[build]   Warnings:  None.                                                                         
[build]   Abandoned: 1 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                            
[build] Runtime: 8.5 seconds total.    
```          
  

in `/home/nrs/path_ws/src/noether/noether_tpp/CMakeLists.txt::77`、
```
    if (${NURBS_FOUND})
      target_sources(${PROJECT_NAME} src/mesh_modifiers/bspline_reconstruction_modifier.cpp)
```
I have no idea to fix this error. 
Any suggestions? Thanks!!

euichan2000, 2024-05-27:
I replaced line 77 of `/home/nrs/path_ws/src/noether/noether_tpp/CMakeLists.txt::77`

before
```
    if (${NURBS_FOUND})
      target_sources(${PROJECT_NAME} src/mesh_modifiers/bspline_reconstruction_modifier.cpp)
```
after
```
    if (${NURBS_FOUND})
      target_sources(${PROJECT_NAME} PRIVATE src/mesh_modifiers/bspline_reconstruction_modifier.cpp)
```

then, above error message disappear. but another error message appear.

```
nrs@nrs-vision:~/path_ws$ catkin build
----------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/nrs/catkin_ws/devel:/opt/ros/noetic
Workspace:                   /home/nrs/path_ws
----------------------------------------------------------------------
Build Space:        [exists] /home/nrs/path_ws/build
Devel Space:        [exists] /home/nrs/path_ws/devel
Install Space:      [unused] /home/nrs/path_ws/install
Log Space:          [exists] /home/nrs/path_ws/logs
Source Space:       [exists] /home/nrs/path_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
----------------------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------------------
[build] Found 4 packages in 0.0 seconds.                                                           
[build] Package table is up to date.                                                               
Starting  >>> ros_industrial_cmake_boilerplate                                                     
Finished  <<< ros_industrial_cmake_boilerplate                [ 0.2 seconds ]                      
Starting  >>> boost_plugin_loader                                                                  
Starting  >>> noether_tpp                                                                          
Finished  <<< boost_plugin_loader                             [ 0.3 seconds ]                      
___________________________________________________________________________________________________
Warnings   << noether_tpp:check /home/nrs/path_ws/logs/noether_tpp/build.check.007.log             
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
cd /home/nrs/path_ws/build/noether_tpp; catkin build --get-env noether_tpp | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

...................................................................................................
___________________________________________________________________________________________________
Errors     << noether_tpp:make /home/nrs/path_ws/logs/noether_tpp/build.make.003.log               
/home/nrs/path_ws/src/noether/noether_tpp/src/mesh_modifiers/bspline_reconstruction_modifier.cpp:40:41: error: return type of out-of-line definition of 'noether::BSplineReconstruction::modify' differs from that in the declaration [clang-diagnostic-error]
pcl::PolygonMesh BSplineReconstruction::modify(const pcl::PolygonMesh& mesh_in) const
                                        ^
/home/nrs/path_ws/src/noether/noether_tpp/include/noether_tpp/mesh_modifiers/bspline_reconstruction_modifier.h:73:33: note: previous declaration is here
  std::vector<pcl::PolygonMesh> modify(const pcl::PolygonMesh& mesh) const override;
                                ^
1 error generated.
Error while processing /home/nrs/path_ws/src/noether/noether_tpp/src/mesh_modifiers/bspline_reconstruction_modifier.cpp.
Found compiler error(s).
make[2]: *** [CMakeFiles/noether_tpp.dir/build.make:492: CMakeFiles/noether_tpp.dir/src/mesh_modifiers/bspline_reconstruction_modifier.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/noether_tpp.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /home/nrs/path_ws/build/noether_tpp; catkin build --get-env noether_tpp | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...................................................................................................
Failed     << noether_tpp:make                                [ Exited with code 2 ]               
Failed    <<< noether_tpp                                     [ 8.3 seconds ]                      
Abandoned <<< noether_gui                                     [ Unrelated job failed ]             
[build] Summary: 2 of 4 packages succeeded.                                                        
[build]   Ignored:   None.                                                                         
[build]   Warnings:  1 packages succeeded with warnings.                                           
[build]   Abandoned: 1 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                            
[build] Runtime: 8.5 seconds total.             
```

marip8, 2024-08-08:
Good catch. I personally (and CI as it happens) don't have a compilation of PCL that includes the NURBS module, so I haven't actually built this file in a while. Your changes should be incorporated in PR #256. Can you confirm if this fixes the issue for you?

marip8, 2024-08-08:
Closing; feel free to re-open if this #256 does not solve your issue

closed 2024-08-08 by marip8