ros-industrial/noether #252: Does Noether support Windows 10?

Dawn-and-Night, 2024-07-30:
I want to configure Noether in the ROS 2 Humble environment on Windows 10. Is this possible?

Dawn-and-Night, 2024-07-31:
> Yes, I'm sure it's possible, but I don't know of anyone who has tried this yet. I'm not a Windows developer myself, so I don't know where the issues might arise. I'm happy to add a CI job for Windows 2019/2022 and accept PRs to help make this repo compatible with Windows though.
Thank you for your response. I am trying to use Noether in a ROS2-Humble environment on Windows 10. My VTK version is 9.1, and my PCL version is 1.12.1. When launching noether_gui_app.exe for path planning, it crashes when using the following tools. If convenient, please assist me. The two images below show the options that cause the crash and the error messages.
[image]
[image]

Dawn-and-Night, 2024-07-31:
Thank you for your response @marip8 .
 I am trying to use Noether in a ROS2-Humble environment on Windows 10. My VTK version is 9.1, and my PCL version is 1.12.1. When launching noether_gui_app.exe for path planning, it crashes when using the following tools. If convenient, please assist me. The two images below show the options that cause the crash and the error messages. 
[image]
[image]

Dawn-and-Night, 2024-08-01:
By rewriting the widget files of these tools that would crash, and modifying the directly using-invoked parts according to the relevant files of the tools that work, everything is now functioning properly. It's quite strange why this issue occurs, as everything works fine on Ubuntu.

Dawn-and-Night, 2024-08-02:
I compiled this using Qt5.15.3 and solved it by making minor modifications to some of the planner's widget files. Taking SnakeOrganicalModifier as an example, referring to other components of the Tool Path Planner that can run normally, I modified its _widget. h and _widget. cpp files, as shown in the figure.
[image]
[image]

marip8, 2024-08-02:
Summarizing in diff format for clarity:

```diff
#pragma once

#include <noether_gui/widgets.h>
#include <noether_tpp/core/tool_path_modifier.h>

namespace noether
{
struct SnakeOrganizationModifierWidget : public ToolPathModifierWidget
{
+ Q_OBJECT
public:
-  using ToolPathModifierWidget::ToolPathModifierWidget;
+  SnakeOrganizationModifierWidget(QWidget* parent = nullptr);

  ToolPathModifier::ConstPtr create() const override;
};

}  // namespace noether
```

```diff
#include <noether_gui/widgets/tool_path_modifiers/snake_organization_modifier_widget.h>
+ #include <noether_gui/utils.h>
#include <noether_tpp/tool_path_modifiers/snake_organization_modifier.h>
+ #include <QFormLayout>
+ #include <QLabel>
+ #include <QDoubleSpinBox>

namespace noether
{
+ SnakeOrganizationModifier::SnakeOrganizationModifier(QWidget* parent)
+ : QWidget(parent)
+ {
+   auto layout = new QFormLayout(this);
+ }

ToolPathModifier::ConstPtr SnakeOrganizationModifierWidget::create() const
{
  return std::make_unique<SnakeOrganizationModifier>();
}

}  // namespace noether
```

Some thoughts:
- Adding the `Q_OBJECT` macro alone doesn't do anything because we are not currently invoking the Qt MOC on any widget files in this repo in the CMakeLists
- Including `noether_gui/utils.h`, `QLabel`, and `QDoubleSpinBox`  seems unnecessary as nothing from those files is used by this particular widget
- I didn't think it was necessary to set a layout in a QWidget that doesn't have any child elements

It would be interesting to find the set of minimal changes that fixes the issue you encountered. It doesn't feel like any of these changes should have fixed it, but I guess it's hard to tell

Dawn-and-Night, 2024-08-02:
Yes, it's really strange, the fact that the original program works well under Ubuntu means that the code is not problematic. It's amazing how this modification works.