ros-industrial/noether #276: Errors colcon building noether 0.0.1: "Could not find a package configuration file provided by   "cmake_common_scripts" with any of the following names:      cmake_common_scripts"

wglockner, 2024-10-09:
Dist: Ubuntu 22.04
ROS2 version: Humble

When running the following lines in terminal:
```bash
source /opt/ros/humble/setup.bash
rosdep install --from-paths src --ignore-src -r -y
colcon build --cmake-args -DTESSERACT_BUILD_FCL=OFF -DTESSERACT_BUILD_RUCKIG=OFF -DBUILD_RENDERING=OFF
```
The following error is thrown:

```bash
--- stderr: vtk_viewer                                                                          
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcmake_common_scripts.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cmake_common_scripts", but CMake did not find one.

  Could not find a package configuration file provided by
  "cmake_common_scripts" with any of the following names:

    cmake_common_scriptsConfig.cmake
    cmake_common_scripts-config.cmake

  Add the installation prefix of "cmake_common_scripts" to CMAKE_PREFIX_PATH
  or set "cmake_common_scripts_DIR" to a directory containing one of the
  above files.  If "cmake_common_scripts" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< vtk_viewer [2.47s, exited with code 1]
--- stderr: noether_msgs
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< noether_msgs [2.50s, exited with code 1]

Summary: 7 packages finished [5.78s]
  2 packages failed: noether_msgs vtk_viewer
  12 packages aborted: behaviortree_cpp boost_plugin_loader btcpp_ros2_interfaces descartes_light industrial_reconstruction_msgs noether_filtering osqp_eigen rviz_polygon_selection_tool snp_isu_msgs snp_msgs tesseract_common tesseract_msgs
  3 packages had stderr output: noether_filtering noether_msgs vtk_viewer
  43 packages not processed

```
Any help would be great

marip8, 2024-10-09:
You probably want to be building something off of the `dev` branch, not the `master` branch, which includes lots of ROS1 things like `noether_msgs`, `vtk_viewer`, etc. I would double check the version/commit hash of `noether` specified in your project (assuming you're building Scan-N-Plan workshop)

marip8, 2025-07-15:
Closing; feel free to re-open if necessary