ros-industrial/noether #285: Unable to generate toolpaths on non-planar surface

srsidd, 2025-01-23:
Is there any requirement / assumption noether is making with regards to input meshes? For planar parts, I can generate toolpaths well however for non planar parts, I run into lots of issues. 

Here is one example of a cylindrical part that noether says has no vertex normals.
[image]
However, when I render the normals in meshlab, I can clearly see the normals.
[image]

Here is another one, where the toolpaths are not uniform or consistent - 
[image]

Is there something noether is expecting that I'm missing?

marip8, 2025-01-23:
> Here is one example of a cylindrical part that noether says has no vertex normals.

Previously, the plane slicer raster planner estimated vertex normals within the planner itself. It turns out that it was useful to break that step out into its own `MeshModifier` for use by other planners. Also it seemed restrictive to re-estimate normals when a mesh could come in with normals that were generated in a better or more application specific way. So in #268, we removed the normal estimation from the tool path planner, and replaced it with a check to make sure the mesh has vertex normals. We also added two mesh modifiers that can perform the normal estimation task. To fix this issue, I would suggest running one of these as the last mesh modifier in the list (preferably `VertexNormalsFromMeshFaces`).

The error message you saw was not particularly useful, so I recently updated the messaging in #283 to make this more informative.

> However, when I render the normals in meshlab, I can clearly see the normals.

I think Meshlab can render normals, but it doesn't necessarily save them into your mesh file unless you specifically ask it too.

> Is there any requirement / assumption noether is making with regards to input meshes? For planar parts, I can generate toolpaths well however for non planar parts, I run into lots of issues.

Yes and no. The plane slicer raster planner was mostly designed for flat-ish shapes, but I have seen cases where the it generates decent raster paths for closed non-planar meshes. @InigoMoreno has been working on some of these use cases. I've been slowly incorporating his changes/suggestions from #216 #217 #218 #219 to help with this

srsidd, 2025-01-24:
Thanks @marip8 , the context definitely helps. I added the `VertexNormalsFromMeshFaces` and received a different error - `Vertex 18808 appears to participate in more faces than exist in the mesh; this mesh likely contains degenerate half-edges.` 

As for the second shape, it doesn't seem to matter, which way I plan the toolpaths, they all show up non-uniform and wonky,

[image]

Is there a different planner that might work better for non planar shapes? What is the recommendation for planning on shapes that are curved / non-symmetric etc.?

marip8, 2025-01-24:
I wonder if this has something to do with the [plane slice raster planner fitting splines to the generated waypoints](https://github.com/ros-industrial/noether/blob/738152873f68980d48047970233fb6b1292f8f4d/noether_tpp/src/tool_path_planners/raster/plane_slicer_raster_planner.cpp#L639). I started a branch to remove this (#271), but I haven't had a chance to finish it. You might test with a rebased version of that branch to see if it performs better. Note, the waypoint ordering may not be correct: that's the thing I need to fix before merging that PR.

> Is there a different planner that might work better for non planar shapes? What is the recommendation for planning on shapes that are curved / non-symmetric etc.?

We don't have any alternative options for that at the moment. I've been working on integrating the [heat method](https://doc.cgal.org/latest/Heat_method_3/index.html) tool path planner [here](https://github.com/marip8/heat_raster/tree/update/noether), which would probably be a good alternative. It's a real WIP at the moment though

srsidd, 2025-01-28:
I tried the branch from that PR, but see no change in behavior for either part unfortunately. The toolpaths are all still wonky on the cylinder and I don't get any toolpaths for the circular part.